universam1/iSpindel #3: universelle Schnittstelle

universam1, 2016-11-24:
Eine universelle Schnittstelle zum Datenaustausch soll festgelegt werden.

Manuel83, 2016-11-24:
Tip top. Kann ich quasi direkt so einbauen.

Einzig würde ich **spindel** in **spindle** (englisch) ändern. Den Wert von **name** würde ich in Anführungszeichen erwarten.

```
{
name: "iSpindel01",
temperature: 20.5, 
angle: 89.5, 
battery: 84
}
```

BernhardSchlegel, 2016-11-24:
Folgende Vorschläge:

- Jede Api sollte eine Versionsnummer in der url haben, damit alte Geräte noch kompatibel sind falls man mal was ändern sollte
- eventuell den datentyp (`JSON` / `CSV`) mit einbinden, falls man mal mehr haben will
- gültiges JSON
- Ergänzunng um Feld `wort` falls man die interne Berechnung der SW nutzen möchte
- CPU ID fände ich noch schick (Name kann ja frei konfiguriert werden)

`http://<IP>:<PORT>/api/v1/[csv|json]/`

Liefert

```
{
   name: "iSpindel01",
   id: "XXXXX-XXXXXX"  
   temperature: 20.5, 
   angle: 89.5, 
   battery: 84,
   wort: 11.8
}
```

universam1, 2016-11-24:
@BernhardSchlegel 
Erkläre bitte genauer was du unter WORT verstehst?

Gültiges json: compressed json würde ich vorziehen da es Daten spart. 

Auch könnte man properties unter Umständen einfach verwerfen wenn nicht unterstützt durch das backend.

Manuel83, 2016-11-24:
Version im Pfad finde ich eine super Idee. Ich würde die API in v1 so klein wie möglich halten. Also Id oder Name. Ist ja quasi das gleiche. 

Wenn die CPU Id eindeutig ist würde ich diese vorziehen. Je weniger Daten der Nutzer an der Spindel konfigurieren muss desto besser.

BernhardSchlegel, 2016-11-25:
mit `wort` meinte ich einfach die Stammwürze

universam1, 2016-11-25:
@Manuel83 
Der Name ist schon vorhanden, die CPU id kann ich liefern!

[image]

Wenn du den Namen verwenden würdest und dynamisch die Spindel erstellst, würde es das coole Feature ergeben dass man sagen kann "IPA_20.11.16_Behaelter1" o.Ä. um Daten für eine bestimmte Sud abzutrennen. 
Oder man nennt ihn "Spindel_Rot" und weiß welche gemeint ist.

@BernhardSchlegel 
OK also optionale zusätzliche Values. Man kann den Standard ja insofern offen lassen dass es Erweiterungen gibt. Mit JSON wäre das ja auch kein Bruch in der Abwärtskompatibilität.

Manuel83, 2016-11-25:
Ja, genau so mache ich das bei CraftBeerPi schon mit den DS18B20 Sensoren. Für jeden kann man in der Software einen Namen frei vergeben. Das mapping übernimmt das Backend.

Manuel83, 2016-12-01:
Wie berechnen die Stammwürze anhand des Winkels? Kannst du mir evtl eine Formel bereitstellen?

So langsam wird es mit dem UI.
[image]

universam1, 2016-12-01:
Das sieht gut aus!

Ich kann dir mal als Beispiel das Polynom nennen welches ich in der aktuellen Kalibrierung erhalte:
`Plato = 0.004415613*(tilt)*(tilt) + 0.120848707*(tilt) - 6.159197377`

Das ist allerdings abhängig von der Spindel, jede Spindel wird sich etwas anders verhalten. Wieviel das ist (abgesehen vom Offset Winkel) kann ich mangels Erfahrung nicht sagen. 
Kannst du das Polynom als Variable hinterlegen lassen?

[image]

Ich habe die HTTP Schnittstelle auch schon implementiert, kann es mangels freiem Raspi momentan nicht testen. Hast du evtl. ein Gateway zum Internet offen an welches ich pushen könnte?

Manuel83, 2016-12-02:
Top Danke. Man kann später für jede Spindel ein eigenes Polynom einstellen. Leider habe ich keinen Server im Internet laufen. Testen lohnt auch noch nicht wirklich. Ich implementiere gerade den "Happy Day" Ablauf. Das ganze muss dann noch robust gemacht werden. 

[image]

Die Variable muss "tilt" heißen.

Aktueller Status:
- Spindel kann sich automatisch anmelden
- Tuning Parameter kann gesetzt werden
- Stammwürze wird berechnet
- Daten werden in einer Datei gespeichert
- Spindel kann einem Gärtank zugeordnet werden

Fehlende Funktion:
- Spindeldaten im Diagramm anzeigen
- Spindel löschen
- Spindeldaten löschen
- CSV Daten Export
- Kühlung und Heizung anhand der Spindeltemperatur steuern

universam1, 2016-12-02:
Richtig gut!
Spannend wie du die Formel parsed!

Für die Zukunft, muss noch die Grundlagen erarbeiten, aber die Temperatur muss hier auch in die Formel, wie bei einer normalen Spindel auch. 

Off topic:
Ich bin umgestiegen auf orange Pi weil billig und stärker. Auch mit gpio soweit kein Problem. Siehst du eine Chance oder prinzipielle Probleme? Frage weil ich kein raspi da habe. 
Merci

Manuel83, 2016-12-02:
Ich meine bei orange Pi war die GPIO Ansteuerung anders. Python wird wohl laufen.

Aktuelle ist es ein sehr simples eval() mit dem ich die Formel evaluiere. Quasi noch eine Sicherheitslücke. Hier kommt entweder noch ein Regex rein das die Formel prüft und Werte vor der Ausführung prüft oder die Formel wird zerlegt und die Parameter werden einzeln konfiguriert. 
Erstmal gehts darum Erfahrung zu sammeln.

universam1, 2016-12-05:
Die generelle Auswahlmöglichkeit verschiedener Backends ist schon mal implementiert. Den generic HTTP habe ich auch, muss aber noch getestet werden.

[image]

Manuel83, 2016-12-05:
Sehr cool!

Manuel83, 2016-12-06:
Ich bin mir nicht sicher wie ich Situation behandeln soll wenn nicht alle 5min Informationen von der Spindel gesendet werden.

[image]

universam1, 2016-12-06:
Bezüglich dem PID, sprich keine Daten für den closed loop?
Evtl. letzte valide Daten für ein paar iterierenden, danach fallback auf Alternativen Sensor?

universam1, 2016-12-07:
@Manuel83 
Ein valider Test wäre dann folgender:

`curl -X POST -H "Content-Type: application/json" -d '{"name": "iSpindel01", "id": "XXXXX-XXXXXX", "temperature": 20.5, "angle": 89.5, "battery": 84}' http://YOURSERVER:PORT/api/v1/json`

Korrekt?

universam1, 2016-12-07:
First implementation of the universal Interface.

Anyone likes to try:
[alpha 3.0.0](https://github.com/universam1/iSpindel/raw/master/bin/alpha-iSpindel.3.0.0.bin)

Henielma, 2016-12-07:
I tried alpha 3.0.0 on a Wemos D1 board with the following logging after a reset. Is it configured right?
I use a local IP address, do I still need to fill in a URL?
```
mounting FS... mounted!
                                                        
reading config file
                                                            
opened config file
                         
                                                        
reading config file
                                                            
opened config file
                                                             

                                                                               
parsed json
                                                                    
parsed config:
                                                                 
{"Name":"iSpindel000","Token":"","Sleep":10,"Server":"192.168.2.17","API":2,"Por
t":80,"URL":"verbeek.rr.nu"}
                                                   
woken from deepsleep, normal mode
                                              
Spl 0: 0.00
                                                                    
Spl 1: 0.00
                                                                    
Spl 2: 0.00
                                                                    
Spl 3: 0.00
                                                                    
Spl 4: 0.00
                                                                    

                                                                               
a: 0	   0	      0	      absTilt: 0.00	  T: 36.53	       V: 0.00OW DISCONNECTED o
r uninitialized: 
                                                              
	       owT: -127.00
                                                           
After waiting 4.40
                                                             
 secs in setup() connection result is 3
                                        
192.168.2.15
                                                                   

                                                                               
sleeping: 10sec
```

The above text is viewed in the logscreen after configuration and a reset. I only use a Wemos D1 board without the accelerator board and the DS18b20. Probably thats the reason why the Wemos board only one time gives the above logging after a reset.

Furthermore I still need to puzzle further on the Labview side to read the postdata.  This does not work yet.

universam1, 2016-12-08:
@Henielma You are missing the 470 Ohm resistor to wake up from sleep

Manuel83, 2016-12-08:
Ja, das wäre ein valider Test. Die Server URL kann man aber später komplett frei eingeben oder?
Weil ich werde den Endpoint wahrscheinlich nicht auf /api/v1/json liegen haben.

universam1, 2016-12-08:
Würde dann dein "CraftBeerPi" als preset URL festlegen. Oder wird die nicht überall identisch sein?
Die Variante "HTTP" wäre dann voll frei konfigurierbar.

Manuel83, 2016-12-08:
Klingt super. Dann muss der Endpoint wie folgt aussehen:
/api/hydrometer/v1/data

universam1, 2016-12-08:
Port: 80?

Manuel83, 2016-12-08:
CraftBeerPi läuft standardmäßig auf Port 5000

Henielma, 2016-12-08:
Thanks for the 470 ohm tip. This resistor added and also the accellerometer board added. The serial screen information does now refresh after waking up. I will test V3.0.0 further because V3.0.3 does not run here. It gives:
```
00001 3ffefe84 3fff02cc                               
3fff1270:  3fffdad0 00000000 00000001 4020c750            
3fff1280:  3fffdad0 00000000 3ffeff8c 3fff02cc  
3fff1290:  3fffdad0 00000000 3ffeff8c 40207928  
3fff12a0:  3fffdad0 3ffeff54 3ffeff8c 3fff02cc  
3fff12b0:  3fffdad0 00000000 3fff0210 4020d036  
3fff12c0:  feefeffe feefeffe feefeffe feefeffe  
3fff12d0:  feefeffe feefeffe feefeffe 3fff02cc  
3fff12e0:  3fffdad0 00000000 3fff02c4 40212e8c  
3fff12f0:  feefeffe feefeffe 3fff02e0 40100718  
<<<stack<<<
                                                                 
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
                                                          
load 0x4010f000, len 1384, room 16 
tail 8                                                
chksum 0x2d                                
csum 0x2d                                         
v00000000                            
~ld                            
     mounting FS... mounted!            
Exception (3):
epc1=0x40215d00 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40255a50 depc=0x00000
000
                                   
ctx: cont                                                     
sp: 3fff0ee0 end: 3fff1300 offset: 01a0
>>>stack>>>                     
3fff1080:  40255a50 00001300 3fff2140 4020354f  
3fff1090:  00008001 000000f8 0000005f 6f632f01  
3fff10a0:  6769666e 6f736a2e 0000006e 00000000  
3fff10b0:  00000000 00000000 00000000 40205b00  
3fff10c0:  00000000 00008001 00000000 00000000  
3fff10d0:  00000000 ffffffff 4020368c 3fff26a0  
3fff10e0:  3fff2140 00000000 00000000 40203a4d  
                              
```

universam1, 2016-12-08:
Thanks thats true, found the issue, will post a fixed version tomorrow

universam1, 2016-12-09:
@Henielma fixed [3.0.4](https://github.com/universam1/iSpindel/tree/master/bin/testing) is out

Henielma, 2016-12-09:
This is what I get in the log screen.

```
FW 3.1.0
mounting FS... mounted!
reading config file
opened config file
parsed json
parsed config:
{"Name":"ISpindel000","Token":"","Sleep":5,"Server":"192.168.2.17","API":1,"Port
":80,"URL":""}
woken from deepsleep, normal mode
Spl 0: 81.27
Spl 1: 81.19
Spl 2: 81.22
Spl 3: 81.24
Spl 4: 81.24
Spl 5: 81.20
Spl 6: 81.15
a: 2170	        14348	  902	    absTilt: 81.22	 T: 28.11	       V: 4.51ERROR: OW
 DISCONNECTED
	       owT: -127.00
After waiting 2.00 secs in setup() connection result is 3
192.168.2.14
 
ERROR genericHTTP: couldnt connect
            
sleeping: 5sec
```
Hereafter the URL setting I configured in Labview.

[image]

No data is read in Labview now. Any ideas what I need to change?

universam1, 2016-12-09:
please select the "HTTP" version and specify the url, port and ip. What happens?

Henielma, 2016-12-09:
I did select HTTP in the configuration of the iSpindel and filled in the next field the IP address of the Tcontrol PC. Strange is that the field under HTTP is called Ubidots Token.

[image]

universam1, 2016-12-09:
hmm your dropdown is not initialized after page load. Okay will fix that. Clicking once the dropdown should fix also.

Henielma, 2016-12-09:
When I first select the other options I can fill in the next fields. I am not sure what to fill in at the URL field in my local network.

[image]

Using the above stettings I get the following logging:
[image]

universam1, 2016-12-09:
in URL you would set "/api/hydrometer/v1/data", the other seam to be ok. I will add the firmware more debug output tomorrow

Manuel83, 2016-12-09:
That's cool. Thanks a lot. 
I committed a version to the branch 
https://github.com/Manuel83/craftbeerpi/tree/spindle

Tomorrow I will add the features to delete the hydrometer and to use temperature values to control the refridgerator

universam1, 2016-12-12:
Unfortunately, no Raspi available. Anyone able to test?

Manuel83, 2016-12-12:
Unfortunately no Hydrometer available to test :-) 
Is it possible that you send me hydrometer? Of course I will pay for it.

universam1, 2016-12-13:
Sure contact me universam@web.de

universam1, 2016-12-13:
3.3.2 has now tested CraftBeerPi support 😃 

[image]

notskumpumpe, 2016-12-18:
Hey Sam & Manuel,

will try to connect iSpindel to CBP . My CPB hangs due to an update at the moment, but I think it will be fixed soon. As I said before: great jobs! Thumbs up. Will report, if the two have been marriaged.

notskumpumpe, 2016-12-18:
Hey Manuel and Sam,

updated my CBP and iSpindel today, how could I test a connection between both ? 

Ahoi, not

Manuel83, 2016-12-18:
[image]
CBPi and iSpindel is working together. I tested today. I will push the code tomorrow to master branch of CBPi

universam1, 2016-12-18:
Awesome- time to celebrate!

Manuel83, 2016-12-20:
Here you will find the CBPi Branch which provides the Hydrometer feature

https://github.com/Manuel83/craftbeerpi/tree/hydrometer

By the way. Does the iSpindel support also Fahrenheit or do I have to calculate this inside of CBPi?

universam1, 2016-12-20:
Great job @Manuel83!!
Looking forward to arriving of my Raspi. 

Currently there is no Fahrenheit support. Do you see a benefit of providing it through the iSpindel?

universam1, 2017-01-25:
Since we have no issues I'm going to close it.

closed 2017-01-25 by universam1