universam1/iSpindel #4: Button zum "History löschen"

JensW2000, 2016-11-27:
Das Config Menü (captive Portal) könnte einen Button zum Löschen der Historydaten bei UbiDots & Co. bekommen, damit man bei neuen Gärungen mit einem leeren Verlauf anfangen kann.
UbiDots stellt dafür z.B. eine DELETE API zur Verfügung.

universam1, 2016-11-27:
Konzeptionelles Problem dass man im CP keinen Internet Zugriff hat. 
Man könnte evtl. ein Flag setzen für den nächsten reset im normal Mode, wenn du Lust hast kann du das ausarbeiten?

universam1, 2016-11-27:
Edit: man kann das sogar in der aktuellen Session machen, da man nach dem CO direkt in normal Mode kommt! Top!

JensW2000, 2016-11-27:
Das Löschen baue ich gerne ein. Ich versuche mal ob ich es hinbekomme. 

Konntest Du den Merge Request jetzt lesen?

 

Gruß

Jens

universam1, 2016-12-07:
Decided this have to happen in higher logic than firmware.

closed 2016-12-07 by universam1