universam1/iSpindel #6: compensate temperature drift of Accelerometer

universam1, 2016-12-05:
MPU 6050 has a temperature drift that needs to be compensated

liuto-attiorbato, 2016-12-06:
Once I have my spindle, I could look into this. I have been working with accelerometers and sensor development for decades.
Unfortunately, this will take a few weeks, as I am still waiting for the omnibus order.
Gernot

universam1, 2016-12-06:
Great, thats nice!

We already do have the internal chip temperature of the MPU available. 
What we would need is the coefficient somehow, the implementation would be very simple.

Unfortunately, I haven't found any documentation about it yet...

liuto-attiorbato, 2016-12-06:
The data sheet says that there is a +-0.02%/K accelerometer sensitivity change and a +-35 mg (X and Y axis) or +-60 mg (Z axis) offset change over the temperature range of 0°C to 70°C. Not a lot, so it might be not worth the effort. We are only dealing with temperatures between - say - 5°C and 25°C, so the sensitivity changes less than +-0,4%
The problem is that we do not know the sign of the coefficient which is probably different for each sensor. I am intending to make three or four spindles and could put these into a climate chamber in my company and measure the drift. Hope Jens is getting some news from the ebay dealers soon.
I also found some discussions in a few forums, mostly dealing with gyro temperature compensation. We'll see.

[MPU-6000-Datasheet1.pdf](https://github.com/universam1/iSpindel/files/634097/MPU-6000-Datasheet1.pdf)

universam1, 2016-12-06:
Thank you for research! The data sheet made not too much sense to me, but thinking twice the +- could also describe a curve here, and we would have to determine the factor that applies to our area.
Probably, your climate chamber would be the best solution at all!

I was also surprised when one user reported an error of 0.8°P because of sunlight, given this is correct it should still be easy to compensate in our use case.

Henielma, 2016-12-08:
Reading the datasheet I see:
1. Enhanced bias and sensitivity temperature stability reduces the need for user calibration
2. Digital-output temperature sensor

According to the first line I'am not sure that additional temperature compensation is needed for the acceleration sensor. There is already temperature compensation on board of the chip. And if we do want a temperature compensation of the acceleration sensor we must use the internal temperature sensor of the acceleration chip otherwise we probably make is worse.

Probably we need another temperature compensation. I mean the "normal" temperature compensation for regular hydrometers. For this compensation we better use the DS18b20 sensor.

universam1, 2016-12-09:
> And if we do want a temperature compensation of the acceleration sensor we must use the internal temperature sensor of the acceleration chip otherwise we probably make is worse.

Correct, we do have this value already, so we can do compensation once we know the coefficient

> Probably we need another temperature compensation. I mean the "normal" temperature compensation for regular hydrometers. For this compensation we better use the DS18b20 sensor.

Absolutely. Since in contrast to the above error this is more a physical influence. This would have to go into this formula [Formula](https://github.com/universam1/iSpindel/blob/master/docs/Calibration_en.md#formula)
It is better there because it strongly depends on the physical dimensions of the actual housing and the calibration temp. the user performed.

Henielma, 2017-01-07:
> 	
> jsled's answer is for when you are using a hydrometer calibrated to 59 degrees F.
> The formula to correct a reading for any hydrometer calibration temperature is as follows:
> cg = corrected gravity
> mg = measured gravity
> tr = temperature at time of reading
> tc = calibration temperature of hydrometer  
> 
> cg = mg * ((1.00130346 - 0.000134722124 * tr + 0.00000204052596 * tr2 - 0.00000000232820948 * tr3) / (1.00130346 - 0.000134722124 * tc + 0.00000204052596 * tc2 - 0.00000000232820948 * tc3))

After googling I [found ](http://homebrew.stackexchange.com/questions/4137/temperature-correction-for-specific-gravity)the above formula to calculate normal hydrometer temperature compensation. This works also for the iSpindel for the physical temperature influence. Keep in mind that the formula's uses temperatures in Fahrenheit. In the iSpindel functionality  of [Tcontrol ](https://sites.google.com/site/henielma/home)I used these formula's.

universam1, 2017-01-09:
Could you @Henielma convert this formula to Celsius since this is available insight in FW?

Henielma, 2017-01-09:
Tcontrol uses this formula but first it converts both temperatures from °C to F. The formula for the conversion of °C to F is: (Temperature * 1.8) + 32. Converting the complete formula to °C is not my cup of tea. What you can do is replace tr in the cg formula by ((tc*1.8)+32) and tr by ((tr*1.8)+32).

These calculations must be done by the applications I assume and not by the iSpindel itself. In Tcontrol I use for the tc temperature the iSpindel temperature as it was during the calibration of the iSpindel. Keep in mind that Tcontrol calibrates the iSpindel tilt just after yeast pitch.

Henielma, 2017-01-10:
When the conversion line from tilt to SG was a straight line I would definitly say yes you can use this formula also for tilt. Now I am not sure but my feeling says exchanging SG with tilt is not completely correct. Currently in Tcontrol I use a lineair conversion from tilt to SG but probably later I need to finetune this lineair conversion.

Henielma, 2017-01-28:
Last week I did a test to determine the needed temperature compensation for the iSpindel I have. The iSpindel did float into sugar dissolved into water and had a SG measured with a normal hydrometer of 1020. I covered the can with foil to prevent evaporation. Tcontrol with iSpindel interface was used to make temperature steps. See the following screen shot for the temperature steps.
[image]
The turquoise curve is the compensated SG / 3 curve compensated using the jled's formula but in the other direction. In practice I found that the iSpindel temperature compensation has an inverted sign comparing to normal hydrometers. Probably this is due to expansion of the petling depending on the temperature.

After this test I used the log file and Excel to see if I could further optimize the temperature compensation.
[image]
The grey curve is the temperature compensated SG using the simple formula: 
SG – (TemperatureCorrectionFactor X (TemperatureTilt – 20 °C)).
For my situation the TemperatureCorrectionFactor 0,18 SG point per °C is the optimum.

The attached Excel file contains the details.
[iSpindel test3.xlsx](https://github.com/universam1/iSpindel/files/737122/iSpindel.test3.xlsx)

In the current version of Tcontrol V065 the jled's temperature compensation is exchanged to the simple 0.18 SG point per °C compensation.

universam1, 2017-05-01:
with new Firmware 5.x the compensation is available over expression evaluation with the variable `temp`

closed 2017-05-01 by universam1