universam1/iSpindel #13: Use DS18S20 instead of DS18B20

jaapz, 2016-12-24:
My supplier sent me the wrong temperature chips, so instead of the DS18B20 I got the DS18S20. This sensor does not support setting the resolution but other than that it should work the same.
However, I'm getting the "OW disconnected" error, which means (i think) that the one wire protocol did not initialise correctly. I commented out the part where the resolution is set and rebuilt the software. I'm still getting the error though...

Do you think this is a hardware problem, or are there other parts of the code that need to be changed to also support the DS18S20 temperature sensor?

jaapz, 2016-12-24:
As far as I can see the sensor should just work fine, the DallasTemperature library doesn't do anything when setting the resolution on this type of sensor... I'm going to re-check my breadboard for any mistakes.

universam1, 2016-12-25:
I think the S20 should work out of the box without modification. But I would highly discourage using it because the lower resolution makes it pretty useless for monitoring fermentation or even more the temperature controlling we do with CBP or TControl.

jaapz, 2016-12-25:
Yeah, I've already ordered the B20 because the 0.5 degrees resolution is pretty bad.
Anyway, I'm getting the OW disconnected error still, and I have checked and re-checked my breadboard, but I couldn't find any problems, the temperature sensor is connected exactly like on the schematics.

I have already tried shortening the data cable, it was about 10cm long so I thought maybe that was too much. The shorter cable is a 2 or 3 cm long but I'm still getting the error.

Do you have any idea where the problem might lie?

Happy holidays by the way.

universam1, 2016-12-25:
I did also have a dead DS18b20 delivered, lost quite some time figuring that out. So just replacing might solve.

jaapz, 2016-12-25:
I tried two different S20's already, but not with the shorter data cable. I'm going to try that later today, thanks.

jaapz, 2016-12-26:
Never mind, looks like when the GY-521 chip is faulty it also affected the initialisation of the 1-Wire bus. I replaced the GY-521 chip and both the gyro and the temperature chip work now.

closed 2016-12-26 by jaapz

universam1, 2016-12-26:
That's very interesting observation, just had the exact same happening with me today...