universam1/iSpindel #15: generic HTTP - optional Header

tkoeberl, 2017-01-08:
Hi,
I would like to send my data to splunk.
in splunk the token has to be set in the header, like in this curl request:
`curl -k  https://localhost:8088/services/collector/event -H "Authorization: Splunk B5A79AAD-D822-46CC-80D1-819F80D7BFB0" -d '{"event": "hello world"}'`
would it be possible to add a header var to the generic http call? 

regards,
-thomas

universam1, 2017-01-09:
Hi, 
shouldn't be a problem. Would require some more generalisation of the generic implementation. I'll think over it.

tkoeberl, 2017-01-09:
Great!

ckrack, 2017-07-20:
+1 The Authorization header is pretty much a standard in Restful APIs.
Maybe you can allow a choice between X-Authorization and Authorization, as both are used.
Usually "vendors" require you to send their name (like "Splunk") and the token, sometimes it's just the token, sometimes it's Username Password. so anything after the header should be fed by a text input, it's up to the vendor to describe how to put this in.

universam1, 2017-07-20:
What about a generic string field `additional header` where one could add whatever he demands?

universam1, 2017-08-16:
no response so closing

closed 2017-08-16 by universam1

tkoeberl, 2017-11-25:
Sorry, missed your comment, 
The "additional header" sounds great.
Can you please reopen the issue?