universam1/iSpindel #16: JSON to TCP Socket (no http)

DottoreTozzi, 2017-01-12:
As we already discussed by email, I'd like to write a small, very basic Python script that receives the iSpindle's data over a TCP socket, without any higher-level application protocol such as http.
The goal is to have a dedicated platform independent server for the spindle, with minimum overhead and battery drain.

The first step will be to simply write the data to a CSV file on a network share for Excel import.
Later I want to expand the capabilities to include database support and perhaps some http server functionality for client-side data retrieval.
The source code will of course be made available.

In order to accomplish this, I suggest the spindle sends a JSON string in the already defined format to a configurable IP Address (host name resolution would be nice-to-have) and TCP port (I suggest 9501 as a default).
After the transmission, the connection could simply be closed by the spindle, or I implent some response (no problem), for example ASCII 6 (ACK), and ASCII 21 (NAK) in case the transmission failed for some reason, perhaps even triggering the spindle to retry for a defined number of times before giving up.

On my side, I'm fully open to suggestions. Whatever works best for you.

Thanks again for your hard work!

DottoreTozzi, 2017-01-13:
I've had quite a productive night shift, so the first preliminary version of my script is done.
It's just ~50 lines of code. My first Python script ever, but I can already tell I quite like it.
Things can be done really fast with it.
Took me just a couple of hours to put this together, even as a complete noob. ;)

Right now, all my script does is listen to port 9501 and send ACK (ASCII 6) if data is received according to specifications, NAK (ASCII 21) if it's not. Excel Timestamps are incorporated in the CSV output per default. Connection won't be dropped by the server by default, however, this would just be a matter of uncommenting a line or two inside the code.

For now, it then just appends the data to a CSV file. 
The filename is generated from "name" and "ID" of the JSON input, so it should be thread-safe under normal circumstances, even if two or more spindles try to connect at the same time.
The TCP listener forks accordingly.

No external libraries are required, it all "just works" in plain vanilla Python 2.7 or later.
Configuration still means editing the file itself. No big deal, though.

I've attached it, so you can play with it when you find some time.

Cheers
Stephan

[iSpindle.py.gz](https://github.com/universam1/iSpindel/files/703574/iSpindle.py.gz)

universam1, 2017-01-13:
Awesome!

Had no time to test your work but I did my part as well! :)

Please try https://github.com/universam1/iSpindel/raw/genericTCP/bin/testing/iSpindel.3.6.0.bin
There is currently no testing against your response, but it's printed to Serial for your debugging. 
You will be able to select as API now "TCP".

Let me know how things go, you can check the code at the "genericTCP" branch.

DottoreTozzi, 2017-01-14:
OK I've had some trouble flashing the firmware from my mac, (Windows inside a VM didn't work) and had to use native Windows in Bootcamp, which worked...

After that, during configuration, the Wemos crashed a few times.
Can't reproduce that now, however, it seems to work now.
Also, switching to TCP mode at first still showed the Ubidots token as input field, only when switching to it for a second time, it showed the ip address and port fields.

Now it works and the good news is I am receiving data.

But:
My script expects the JSON to come in one line.
Looks like the spindle sends the leading "{" first, then the rest of the JSON.
This could of course also be just due to my q&d implementation.

Here's my debug output:
`waiting for connection... listening on port: 9501
('192.168.2.28', 25603) received:'{'
('192.168.2.28', 25603) Error: Expecting object: line 1 column 1 (char 0)
('192.168.2.28', 25603) NAK sent. Waiting for Retry.
('192.168.2.28', 25603) received:'"name":"iSpindel01","ID":"14236088","angle":31.09,"temperature":25.25,"battery":4.30}'
('192.168.2.28', 25603) Error: Extra data: line 1 column 7 - line 1 column 86 (char 6 - 85)
('192.168.2.28', 25603) NAK sent. Waiting for Retry.`

DottoreTozzi, 2017-01-14:
Never mind, I've fixed it.
It will now gather incoming data until it encounters "}" (as it should have in the first place).
The socket is being closed after that (or when it times out).
[iSpindle.py.gz](https://github.com/universam1/iSpindel/files/705742/iSpindle.py.gz)

Works like a charm now. Might want to take a look at the other issues I've mentioned before, though.

Latest version of my tiny lil' script:
[iSpindle.py.gz](https://github.com/universam1/iSpindel/files/705745/iSpindle.py.gz)

universam1, 2017-01-14:
Great I will to cache the tcp data before sending if that makes a difference. 
Also I wonder what a close signal should be sent. 
It is not closing because it is waiting for the reply of your script, I guess that's the reason. Can you ack and close?

DottoreTozzi, 2017-01-14:
Yes that's what I've done now. I send ACK or NAK and close the connection server side.

universam1, 2017-01-14:
Nice, do you mind to add your script and a bit description to the Repo?

DottoreTozzi, 2017-01-14:
Sure thing! Just been waiting for your OK to do so. Where should I put it?

universam1, 2017-01-16:
Great!
What about this folder https://github.com/universam1/iSpindel/tree/master/tools maybe make a subfolder and add some description in `/docs` folder?
Thanks

DottoreTozzi, 2017-01-17:
Done. Sorry for spamming Pull Requests, I've worked with GIT before but sometimes its features are still a bit elusive... (ashamed smily here)...

universam1, 2017-01-17:
Thats just perfect! Thanks so much, have to link to your work now in the Readme's. If there's something more to do feel free to open.

closed 2017-01-17 by universam1