universam1/iSpindel #21: 230K Ohm resistor is not a common value?

Calimerorulez, 2017-01-26:
Hi,

I'm trying to find a suitable resistor for the 230K one, but it isn't a common value. Can I use another value like 229K or 232K ?

universam1, 2017-01-26:
for sure, I introduced a configuration option in the UI to adjust your changed conversion factor. And it is simple like `measured volt/displayed volt * current factor = your new factor`

Calimerorulez, 2017-01-26:
Thank you for your quick reply! :-)

closed 2017-01-26 by Calimerorulez

tedelm, 2018-02-14:
So, with a mesured volt of 4.2
Displayed volt of 4.1
220kOhm resistor

(4.2/4.1)*220000=?

Or am I missing something?

tedelm, 2018-02-14:
Or is it
(measured volt/displayed volt) * current conversion factor = new conversion factor

default conversion factor is 200

universam1, 2018-02-14:
This is covered in the docs FAQ