universam1/iSpindel #27: pet preform

ggazzo, 2017-02-13:
I can't find any pet preform like as suggested in my country, is there some place to buy one ( like ebay, aliexpress, etc) ?

is secure use another size/shape?

thanks.

universam1, 2017-02-13:
The PET preform is about 15x4cm. Its a difficult part to find the correct size to meet the boundaries of the operation. On this mentioned volumina the overall weight should meet pretty exactly 200g to be at 25° tilt in pure water.
If you find something similar let me know. In Germany these preforms are likely available for geocachers, maybe that helps.

ggazzo, 2017-02-13:
something like [that](http://www.ebay.com/itm/5-Brown-PET-Micro-Geocaching-container-geocache-Petling-preform-soda-waterproof-/142254409848?hash=item211f060878:g:NgUAAOSwNSxU4Hds) ?

this "tip" `geocachers` helps a lot.

As you said, the exact size is a little hard to find.  Do you have some link from Germany? Just to compare and maybe ask to some friend to bring to me.

universam1, 2017-02-13:
sure, take a look into https://github.com/universam1/iSpindel/blob/master/docs/Parts.md documenation there is a seller mentioned.

ggazzo, 2017-02-13:
thanks, actually I'm not so used to read in German. But now, as you show me, I found. Thanks a lot

universam1, 2017-02-13:
have to translate indeed, working on it asap.

closed 2017-02-20 by universam1