universam1/iSpindel #39: Pelting XL dimensions

alscheuring, 2017-03-04:
Hi,
Can you provide the dimensions (size and volume) of the Pelting XL? The website doesn't have them in stock and I was hoping to source from somewhere else. If I find a suitable replacement I'll help add it to your documentation.

pricelessbrewing, 2017-03-04:
From the sellers page: 
diameter: ~4cm
Height: ~15cm

Assuming those are the interior dimension, and not the exterior dimensions, that puts it at ~0.188L. It's likely 150mL or 175mL though, as that's an atypical size. Perhaps someone that owns the pelting already can measure theirs.

universam1, 2017-03-07:
Measured inside dimensions are

- length 15cm
- diameter 3.9cm

You may need to find a similar PET Preform also in your country but it's indeed not an easy task. If you find something similar with these dimensions the overall weight is ~200g.

Hope that helps
Sam

closed 2017-03-09 by universam1

beikeland, 2020-02-11:
I know this is a old issue, but having tried to source the Petlings recently I thought I'd add some up to date metrics.

eBay seller geocachingshop seems to sell a size similar to "original" (eBay item 283418077831). ID at the top is 39mm, internal length is 148mm + 11mm for the rounded tip. The ID tapers down to maybe 37mm towards the 148mm mark and then  is rounded out at about 20mm internal radius. Outside length without cap is  162mm.  This is sold as XXL.

another eBay seller uniquegeocaches (eBay item 132379354446) sells a smaller one. ID at the top seems to be 33mm. Outside length wihtout cap is 132mm. Seems the taper is mostly on the OD. Sold as XL.