universam1/iSpindel #42: iSpindel does not connect to CraftBeerPi

chipmunk03, 2017-03-09:
Hi, 

I am trying to get ispindel to connect to craftbeerpi. In the Arduino console I see it send data, but cannot connect:
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: MiSpindleOne
*WM: Parameter
*WM: sleep
*WM: 3
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 2
*WM: Parameter
*WM: token
*WM: 
*WM: Parameter
*WM: server
*WM: 192.168.3.33
*WM: Parameter
*WM: port
*WM: 5000
*WM: Parameter
*WM: url
*WM: 
*WM: Sent wifi save page
ERROR: OW DISCONNECTED

Exception (9):
epc1=0x401049a5 epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys 
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  00000056 00000001 4010410a 3ffee670  
3fffff60:  4022b38d 3ffeef68 00000001 00000000  
3fffff70:  4022666f 00000004 00000000 00000000  
3fffff80:  3fff246c 40226066 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffea578 00000000 4023b563  
3fffffa0:  3ffea578 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v04df3adb
~ld

FW 4.1.7
2.0.0(656edbf)
Worker run!
Boot-Mode: 2
mounting FS... mounted!
reading config file
opened config file

parsed json
parsed config:
{"Name":"MiSpindleOne","Token":"","Sleep":3,"Server":"192.168.3.33","API":2,"Port":5000,"URL":"","Vfact":191.80,"SSID":"***********","PSK":"***********"}...........
ERROR no Wifi credentials

going to Config Mode
{d l��| �l�|�l�c|����{�cd�b��gg�lgo���cp��drlrd�s�g�c�ocx �

I can make a connection to my RPi using TCP. I've got that working, but using the craftbeerpi setting it is not working. I build the firmware using Arduino 1.8.1 on Linux (Elementary OS). iSpindel is latest pull from github.

I tried to configure from a tablet, laptop and smartphone. No success. Same error. TCP works fine.

universam1, 2017-03-09:
You would need to analyze your stack trace, there is an howto on the esp/arduino project. I cant analyze it since you are using your own compiled version.

universam1, 2017-03-09:
Also get your temperature sensor working before digging deeper, might be related.

closed 2017-03-09 by universam1

chipmunk03, 2017-03-09:
Hi,

I took the Wemos board off the the "main" board, so it could not "see" the DS18B20. With the wemos board back on the main board, it sensed the DS18B20. 

Here is the stack trace. I hope it can be of any help:

>>>stack>>>
3fff1170:  00000009 00000000 00000002 4020df7c  
3fff1180:  3fff016c 00000000 3fff016c 3fff1200  
3fff1190:  3fff016c 0000000a 00000003 40213216  
3fff11a0:  00000009 00000000 3ffe8dc3 40215848  
3fff11b0:  00000002 00000000 00000001 00000033  
3fff11c0:  3ffe8dc3 3ffeff5e 3fff016c 4021301d  
3fff11d0:  3ffe8dc3 3fff1200 3ffeff5e 00000033  
3fff11e0:  3ffe8dc3 3ffeff5e 3fff016c 40213030  
3fff11f0:  3ffe8dc3 3ffeff5e 3fff016c 4020e374  
3fff1200:  3ffe9a48 9f03a8c0 feefeffe 00000000  
3fff1210:  feefeffe feefeffe feefeffe feefeffe  
3fff1220:  feefeffe feefeffe feefeffe 3fff0228  
3fff1230:  3fffdad0 00000000 3fff0220 402141fc  
3fff1240:  feefeffe feefeffe 3fff0230 4010070c  
<<<stack<<<

Decoding 8 results
0x4020df7c: uploadData(unsigned char) at /home/frank/Projects/Arduino/iSpindel/iSpindel00/iSpindel00.ino line 357
0x40213216: Print::print(unsigned long, int) at /usr/local/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40215848: Print::write(unsigned char const*, unsigned int) at /usr/local/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x4021301d: Print::write(char const*) at /usr/local/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x40213030: Print::print(char const*) at /usr/local/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 87
0x4020e374: setup at /home/frank/Projects/Arduino/iSpindel/iSpindel00/iSpindel00.ino line 676
0x402141fc: loop_wrapper at /usr/local/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /usr/local/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109

I also found in Globals.h 'CraftbeepPi'. Shouldn't that be 'CraftbeerPi'? Is it a typo? Or is it on purpose?

I changed it in the code (iSpindel00.ino and Globals.h). That's all I changed in the code. The rest is as is from the git pull request. 

Cheers,

Frank

Ins0x, 2018-11-23:
Hi guys, I am running CBPI 3.0 but I can not get iSpindel conected. It shows 0 SG, 0 V and 0 °C. Any ideas what can be wrong?