universam1/iSpindel #45: Add a README.md to explain the purpose of the different drawer designs

dsvensson, 2017-03-13:
Add a README.md to explain the purpose of the different drawer designs found here:

https://github.com/universam1/iSpindel/tree/master/drawer

Drawer-cachersworldXLv12.stl has a version number, and the highest, but Drawer-combo-short.stl was most recently changed. They look slightly different while Drawer-cachersworldXLv12.stl and Drawer-cachersworldXLv11.stl are indistinguishable to me. Some description is in place.

It looks like the kit here:
https://3d-mechatronics.de/en/ispindel-diy-set-komplett_151.html

Has either Drawer-cachersworldXLv11.stl or Drawer-cachersworldXLv12.stl

dr3x, 2017-03-29:
[image]

The sled is slightly too large for the Ebay container that I bought; image shows how far it goes before it jams.  

**XXLBIG PET Micro Geocaching container geocache Petling preform Soda bottle FTF**
[http://www.ebay.com/itm/142295113730?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT](url)
https://raw.githubusercontent.com/universam1/iSpindel/master/drawer/drawer-combo-short.stl

universam1, 2017-03-29:
Thats indeed the tough part, finding a suitable container.

But here is the trick, put the sled into hot water ~140°F where it gets soft, then push it gently into the sled. Repeat it if necessary, you will squeeze it until it fits. 
Not perfect but works!

-Sam

dsvensson, 2017-03-30:
So based on the above comments this is the only one that should remain in that directory:

https://raw.githubusercontent.com/universam1/iSpindel/master/drawer/drawer-combo-short.stl

and the other ones should be deleted? Is that correct @universam1 ?

universam1, 2017-03-30:
Yeah I know, maybe I should describe the dimensions. Actually since there suppliers PET preform changed its dimensions over the last months, I just left the different sizes there...

I will add dimensions so one could find a fitting sled.

dr3x, 2017-03-30:
I tried the hot water trick, put water in a ceramic dish, floated the sled on the water and put it in the microwave on the beverage setting and it worked well to soften up the material so it could fit the tube, however the sled legs splayed out a little (warped) so flipped it over and pressed it back into shape and that seems to have fixed that issue. I'll find out when I get the circuit board but shouldn't be too difficult to reform it.

dsvensson, 2017-03-31:
@universam1 any idea what design matches this one:

http://www.ebay.com/itm/132118452191

universam1, 2017-04-12:
I've added the length of the different designs. So one might check which of these do fit to his preform. Still you can squeeze that PLA quite easily with hot air or water if necessary.

closed 2017-04-17 by universam1

dsvensson, 2017-04-27:
Is that outer or inner dimensions?

If I measure my container I get:

* Inner:
  * Width: 33.3mm
  * Height: 129.6mm
* Outer:
  * Width-Top-By-Opening: 36mm
  * Width-Middle-Of-Height: 39.2mm
  * Height: 133mm

So I'm not sure which one to pick here:

* drawer-36x133-thing2231986.stl
* drawer-38x133.stl

Maybe a README.md with where to measure would be a nice addition. I don't have a 3D-printer myself so I hope to avoid an extra print-request from my 3D-printing-contact.

universam1, 2017-04-28:
It is the outer dia of the sled hence the inner of the container.

Your container is quite a bit thinner, I guess you have to hack it a lot, it might be really super tight if possible at all.

gh112, 2018-02-07:
I need some help to choose a drawing. (first time 3d printing)

The PET I am going to order is 40.5x149 mm (inner diameter)
I am guessing that i need drawer-38x150-Geocaching.stl and maybe use some sandpaper, or possibly drawer-38x145-Geocaching.stl

Please advise if another one is more suitable

einaruk, 2018-03-06:
Hi @dsvensson 
Were you able to fit a drawer to your 33x130 tube and made it work? I am having difficulties with a quite smaller tube.