universam1/iSpindel #47: Docs unclear for wire going from wemos to lipo

BigValen, 2017-03-17:
"Connect Wemos USB + 5V (above diode pin) to Lipomodul + IN (optionally to load over the Wemos)" is unclear. Does this mean one side of the removed​ diode is to be connected to the lipo ? If so, which side ?

This wire is not shown on https://github.com/universam1/iSpindel/blob/master/docs/iSpindelbreadboard.md which is unfortunate, as my iSpindel is assembled without it :-)

universam1, 2017-03-19:
I think I am going to remove this hint at all. Initially I was using the Wemos USB port also to power the Lipo charge module. I find though that this soldering over strains majority and is not really needed with the charger module having its own port.
Nevertheless, don't forget to remove that very diode, otherwise you are powering the battery directly by 5V and might lead to a nice fire in your house...

BigValen, 2017-03-19:
Ah, good to know. I've mostly got it working now, except the tilt always
returns 0. Will debug my soldering later today.

 When you say "remove the diode", it's probably always good to tell people
why. The feel better about taking a knife to their hardware then :)

jaapz, 2017-03-21:
@BigValen when your tilt always returns 0 the GY-521 chip is probably broken. I had the same issue with some chips I got from china.

kbaggen, 2017-05-14:
Hi there, removed the diode, but wemos is no longer acceasibly from PC over usb? Is it broken then or?
Worked over USB before! Some other guy have same issue! Any hint welcome :-)

universam1, 2017-05-17:
You need to power the Wemos over Battery now since you removed the USB feed. Charge your battery completely otherwise you might have issues to flash.

closed 2017-06-22 by universam1

gesselergaragenbraeu, 2025-02-28:
Hi there

I bought about 2 years ago a ready assembled ISpindel – unfortunately it stops working near the end of last fermentation. I recognized that the battery holder solder connection had come loose on + … so I took out battery and soldered it new. but it not started to work anymore. I would like to repair it and bought a new Wemos D1 original lolin V4.0 and a new LiPo charging board TP4056.

But from the Wemos I have to take out the input diode … the V4.0 looks completly different – unfortunately I could not identify and it is not possible to add a picture here. Is here somebody who can help me in this question?

Thank you very much.

I am more hobby brewer than electronic expert.

Best Regards
Michael

ErikdBr, 2025-03-01:
I have build like 40 of these things and never had any issues. Just make sure you charge the battery thru the TP4056 with the switch in the off-position (and I am not even sure if you have to put it in the off-postition while charging).
The original iSpindel had the USB port of the Wemos facing the lid of the tube.
If you use a board like the The Jeffrey 2.69 the antenna is facing the lid for better WiFi connectivity and the USB port of the wemos is facing towards the bottom of the tube while the USB port of the TP4056 is facing the lid so less room for error.

gesselergaragenbraeu, 2025-03-14:
Hi Erik,

Thank you very much for your answer.

So I unsoldered the Wemos and the TP4056 and soldered a new one... a original Lollin 4.0 ... Firmware flashing from Linux was successfull (without insert the Battery).
I can reach configuration Website by WLAN as long the USB ON the Wemos ist connected and supplies energy. I am able to configurate the Wemos. I get Data about Tilt an Temperature. But... If i disconnect and insert the battery it is dead. Charging by the TP4056 seams to Work LED is blinking and stops .. i beleive when fully charged. I am going to Check out next the battery and the battery-holder... May it is a problem with the energy supply.

Error finding ist not easy for such beginner Like me... I normaly want to brew...

Thank you for your help.

Best regards 
Michael 

⁣BlueMail for Android herunterladen ​

ErikdBr, 2025-03-14:
> Error finding ist not easy for such beginner Like me... I normaly want to brew...

You can always do it the old fashioned way without the iSpindel ;-)

commit 2454a420 (2017-06-29): resolves #47