universam1/iSpindel #59: Deep sleep unreliable with resistor

pronkster, 2017-04-26:
Hi there,

Firstly wanted to say that you are doing an awesome job here. :+1: 

My iSpindle is setup with the **Wemos D1 Mini PRO** which works fine except, the iSpindle wouldn't wake up from deep sleep with the config as described in your Wiki. A quick google told me that this is a known issue with the Wemos in general (all models) and that replacing the 470 ohms resistor with a diode solves this issue. 

Credits go to **schufti** in this forum post: http://www.esp8266.com/viewtopic.php?f=13&t=7918&start=24#sthash.9AUQQ49G.dpuf

> " instead of experimenting with resistors and relying on the "it worked for me" value, it would be more technically correct to use a schottky diode (e.g. BAT43) to cleanly pull rst low without side effects when gpio16 is high.
> 
> p.s.: the cathode (ring) towards gpio16 ..."
> 

It worked for me so thought id share and hopefully it works for others as well.

Cheers

universam1, 2017-04-26:
Thats indeed a great idea! Let me know how it goes! 
Regarding the resistors I found that sometimes a lower resistor ~370 Ohm works better.

pronkster, 2017-04-26:
So far so good! All functions work perfectly, even the reset! No need for a resistor I reckon, just use a diode instead. Needs more testing on the normal Wemos D1 mini as well though

universam1, 2017-04-26:
It seems to me that the selection of the diode to be a schottky is a critical one here!
According to the datasheet, the min. Input Low Voltage must be < 0.25×VIO while the Output Low Voltage is up to 0.1×VIO.
Therefore,  this leaves 500mV where the BAT43 will drop around 300mV or more. So a regular silicium diode will fail in this setup!
Using a BAT43 or better is a requirement so to speak.

pronkster, 2017-04-27:
Are you able to share a link to that data sheet? I have looked everywhere but haven't been able to find a good detailed one yet.

I have replaced the resistor with a normal 100v 1a diode and this seems to work just fine (no BAT43). Im trying to figure out why there is a resistor required but all I can find in the original documents is a direct connection RST to D0.. to me it seems D0 just need to have power for the unit to come back from deep sleep. Would it not be that the lower the voltage the less reliable the unit becomes? Not sure why a diode would work better than a direct connection though..

universam1, 2017-04-27:
The link: http://download.arduino.org/products/UNOWIFI/0A-ESP8266-Datasheet-EN-v4.3.pdf

If it works on your *current* setup it does not imply that it will be reliable under different conditions like a nearly empty battery or at lower temperatures. According to the DS it violates the limits with a normal diode.
The resistor is there to protect the D0 output if through UART chip the RST is pulled down at the same time. I've fried already D1 because of this.

pronkster, 2017-05-02:
Cheers, thanks for the link! I will do some more testing over the next few days and let you know how I go.

Kloorob, 2017-05-09:
Hi,
I have just found this article on Github, as I am having the same issue with my iSpindle setup using a Wemos D1 Mini, which like pronkster works fine when first connecting / linking to Ubidots (all reading show up) then nothing after that, and hence I suspected the iSpindle wouldn't wake up to transmit the next signal / link to Ubidots?

Therefore, can you propose a remedy for this issue please, as I view this project as too fantastic for me to give up on it?

Also, thanks for all your hard work.

universam1, 2017-05-09:
Use a Schottky diode or a lower resistor of 350 Ohms.

pronkster, 2017-05-09:
I finally had some time to look at this in more detail. I found there is a lot of conflicting information online around the use of resistors and diodes and in particular, which one is the correct one (and how do you know for sure). This is what I have found so far: 
  
There are three ways to revive from (deep) sleep;  
1.	You can press the reset button 
2.	You connect RST and GND  with a switch (effectively the same as a RST button) 
3.	You can use the inbuild sleep function  

The inbuild Deep sleep function uses GPIO16 as an Output which is ether HIGH (3.3 V)  during sleep or LOW (0.0V) once the timer has finished. 
 
Connecting D0 (GPIO16) directly to RST as per original manual will result in a constant 3.3 V on RST while in sleep and prevents a manual or software reset to pull LOW. Uploading firmware is apparently not working either (not tested this myself).  

To fix these issues you can either use a resistor lower than 350 Ohm or better, a diode between RST and D0 (pointing to D0 ). 
 
A diode or resistor will prevent 3.3 V on RST while D0 is HIGH during deep sleep thus allowing a manual RST through USB-TTL or the RST circuit. It is also preventing damaging the chip if the RST is pulled down through UART chip when booting up. Apparently the problem is that during during Linux Kernel boot phases 1-3 all pins are floating (GPU to CPU handover) and on Phase 4, the PINs are set to the correct state. Connecting D0 and RST directly can result in a fired WEMOS.  

Using a higher resistor might work when your battery is full but might not work once your battery is running lower so it might be better to use a diode with the lowest amount of internal resistance.  I am using a normal diode (not a Schottky) and this works great so far. However, I have not done any testing over a longer period. I will do soon and update once I know more.

Kloorob, 2017-05-10:
Hi Ponkster.

Hope you don’t mind me asking, can you kindly give me the actual diode part number / type you are using please, so that I can try and use the same and test it on my pcb please?

Thanks.

wizbrewery, 2017-05-11:
Hi,  getting the same issue here using weemos mini pro.  I get a couple of pings one programmed and reset then goes to sleep forever.  Will try the diode trick tomorrow,  cheers

pronkster, 2017-05-11:
@Kloorob - I am using a 100v 1A diode. I believe the product number is 1N4934 but will confirm when I am  home this weekend.

wizbrewery, 2017-05-12:
Hi,  tried with IN4001 thanks @pronkster works great :)

pronkster, 2017-05-13:
Just checked and I am using the 1N4934

universam1, 2017-05-17:
don't use a standard diode because it is out of spec. Use a schottky instead!

pronkster, 2017-05-17:
Hi @universam1, are you able to elaborate on that? Please correct me if I am wrong but my findings are that for the purpose of this project, any diode will do. The Deep Sleep function uses GPIO16 as an output, not as an input so it doesn't really matter if LOW is higher than 0V. The whole purpose of this diode is to protected RST when GPIO16 (or D0) is high during reset or when using the USB firmware upload.

universam1, 2017-05-17:
@pronkster Check my above elaboration  on the datasheet why it is critical to use the right kind of diode here

Phrozentech, 2017-05-19:
@universam1 so using a BAT41 or BAT43 would be the best diode to utilize instead of the 370ohm resistor in this case? I'd like to use something that's consistent, since it looks like if my battery gets below 3.6-3.5 the device doesn't wake up, I'm still testing that theory though.

Kloorob, 2017-05-20:
Hi @universam1. Just to let you know that I built a new board today and added a BAT43 Schottky Diode in place of the 470 ohm resistor, and my iSpindel is now working, which is brilliant!
Thanks for your help.
Cheers.

lekrom, 2017-05-26:
Dear @universam1,
I absolutely agree with your assesment on the diode.
May i recommend you update the docs / circuit diagram to use the schottky diode instead?
Regards,
Lekrom

lekrom, 2017-05-26:
Epic project by the way...

universam1, 2017-05-26:
thanks @Kloorob @lekrom thats good feedback!
Would like a little more feedback but I tend towards this being the better way.

lekrom, 2017-05-26:
A simple yet elegant sollution. Cant think of a better way. Should work for virtually any condition. Schottkys are also typically much faster that Si-diodes, ideal for protection purposes, so should work very well for protecting the IO pin in thet scenario nentioned. Must admit, I was puzzled by the resistor in the circuit. A few quick searches led to this thread. I have been working (and playing) with microcontrollers for many years,  although Esp8266 still very new to me. Missed the wake from deep sleep GPIO16 thing...

Kloorob, 2017-05-26:
Hi @universam1, Just to give you some further background as I have built 5 boards and the first 4 board, I thought that I messed up in the soldering, and hence, I couldn't get it to work. However, on my fifth board, I used a BAT43 Schottky Diode and it worked.  All I need to do now is to put it into a fermenter to give it long test.  
Thanks for your help.

universam1, 2017-05-29:
nice @Kloorob @lekrom I'm fine with it. Feel free to create a Merge Request to the docs, BOM etc. Thanks

Phrozentech, 2017-06-03:
I used a BAT46 instead of a BAT43 and was able to get readings until 3.67v then it stopped communicating. Looks like a 192hrs between diode and using a BAT46 currently recharging the battery and rerunning the test again.

universam1, 2017-06-04:
That's actually a pretty disastrous result,with resistors I achieve stable run until 2.8V down. Once I got some time have to simulate this on a PSU.

Phrozentech, 2017-06-04:
Good to know, so I should be able to get to 3.3v since that should be the cut off on a 18650 to keep from over discharging

pronkster, 2017-06-04:
I am still running with the silicium diode (1N4934) but will go for Schottsky in my next version as that seems to be the more efficient diode choice. On this setup the deep sleep stops working at around 3.28V. 

@universam1; 2.8V is a much better result! Is that with 370 Ohm resistor? Im keen to see your findings running this on a PSU

Alchomist, 2017-06-20:
Not sure if this is relevant to this thread, but I found I could log into the configuration but saving the settings resulted in an error & connection closed. One of the guys at Aussie Home Brewer noted he had the same problem & the BAT43 solved this.

Appreciate your effort on this project universam1

Alchomist, 2017-06-22:
Installed the shottky but it still won't save wifi details or log-in. The Wemos can see 6 wifi signals/extensions but can't log into any of them. It keeps saying error config corrupted then goes into config mode. Weird how it can read temp, tilt &amp; battery &amp; transmit it to the monitor, then get a wifi error &amp; shut down.

Does it sound like a dodgy soldering issue?

universam1, 2017-06-22:
are you having spaces / special character in your wifi password?

Alchomist, 2017-06-22:
No, I changed the wifi password to all lowercase and numbers

universam1, 2017-06-22:
this is not related to this topic, please open a new issue and post your debug terminal output as well as informations about your built.

padrino-xl, 2017-06-26:
Hi there,

thank you for this great project! 
After building my first ispindel and a lot of reading I still have a Problem.

The ispindel sends the values just one time after pressing the reset button and doesn't wake up afterwards.

I unterstand this is relatet to the 470ohm resistor. I tried a 390 ohm resistor and i did not work from the beginning. After a lot of tries it suddenly worked. 
I thought changing to a 350 ohm resistor would make it more reliable. But now it stops again after one time sending.

I read the schottky diode might help. Would you recommend it since it seems to work but not with low battery?
I'll check my soldering again, but I'm pretty sure thats not the Problem.

Thanks again for your great work here!

universam1, 2017-06-26:
Could you please set the timeout to 15s and post here the output of the serial console, I am sure this will shed some light into what's going on here.

padrino-xl, 2017-06-26:
Hi,

this is what I can read:

> {d[00]dœŸ|[00]Œdà<[03][04][04][0C]Œ[0C]lä[04]c|ƒ[02]ì[13]“{›cŒ[0C]cŒóooŸlggœãì[04]c[1C]xŒÇl;lslxògà[10][03][04][04]ƒ[0C]l[0C]„œ[04][0C][0C]c[0C]gã|[03]dì[0C][0C]Ž#„ûooî[00]lŒd`[03]˜[12][13]'n[04]$`[03][07][02]gsŽ››o[0C][0C]c[0C]œ[07]l[0F]{’’g[04][04]c[04]œ[0E]dœ[03][04][04][04][0C][0C][0C][0C]l`[03]üƒnœ[03]
> 
> FW
> 
>  5.1.2
> 2.0.0(656edbf)
> Worker run!
> Boot-Mode: 5
> mounting FS... mounted!
> reading config file
> opened config file
> 
> parsed json
> applying offsets
> parsed config:
> {"Name":"iSpindel","Token":"XXX","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":196.43,"SSID":"XXX","PSK":"XXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":73,"aY":-781,"aZ":883}
> woken from deepsleep, normal mode
> Spl 0: 87.08
> Spl 1: 86.98
> Spl 2: 87.46
> Spl 3: 86.82
> Spl 4: 87.04
> Spl 5: 86.78
> Spl 6: 86.63
> 
> a: 982	16414	100	absTilt: 86.95	T: 31.16	V: 4.11	owT: 29.37	Gravity: 32.03
> After waiting 0.00 s, result 3
> 192.168.0.157
> UBIDOTS: posting
> POST /api/v1.6/devices/iSpindel?token=XXX HTTP/1.1
> Host: things.ubidots.com
> User-Agent: iSpindel
> Connection: close
> Content-Type: application/json
> Content-Length: 65
> 
> {"tilt":86.95,"temperature":29.37,"battery":4.11,"gravity":32.03}HTTP/1.1 200 OK
> Server: nginx
> Date: Mon, 26 Jun 2017 17:52:56 GMT
> Content-Type: application/json
> Transfer-Encoding: chunked
> Connection: close
> Vary: Accept-Encoding
> Vary: Accept, Cookie
> Allow: POST, OPTIONS
> 
> 8d
> {"battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "gravity": [{"status_code": 201}]}
> 0
> 
> 
> Final-sleep: 15s; RT:2320
> rl[00]lœŸ|[00]Œlà|[03][0C][0C][0C]Œ[04]dä[04]c<‡ƒ[03]ä[1B]›{ÛblŽb„ûogŸl'nœãì[0C]c[1C]xŒdslsd[1B]ƒsó’oàcâgcp[00]ü

I managed to change the battery-value, but failed the offset-Calibration. After pressing calibration, the spindel never shows up again and I have to reset again.

universam1, 2017-06-26:
Interesting!
Could you do as a test short wire the d0 and RST and monitor if this changes? Just make sure not to flash firmware while this pins are bridged. This way you will know if it is related to the resistor at all.

padrino-xl, 2017-06-28:
hi,

I couldn't test it yesterday, but here are the readings with the d0 and RST bridged.
It looks like it is working this way. 
I don't know what to take from this. Should I use a diode, or a smaller resistor? At the moment I am using a 350 ohm resistor.

> sd[00]lœŸ|[00]„là|[02][0C][04][04]„[04]lì[0C]c|Žƒ[03]ì[12]’s“c„[0C]cŒógnŸ$ogœãä[04]b[1C]pŒ‡l{d;lpûgà[18][03][04][04]ƒ[0C]l[0C]Œœ[04][04][0C]c[0C]gã|[03]dì[0C][0C]Çc„ûooî[00]lŒŽd`[03]˜[12][13]'n[04]$`[02][07][03]gsŽ››o[0C][0C]c[04]œ[07]l[0F];’“g[04][04]c[0C]œ[0F]dœ[03][04][04][0C][0C][0C][0C][0C]d`[03]ü‚gœ[03]
> FW 5.1.2
> 2.0.0(656edbf)
> Worker run!
> Boot-Mode: 5
> mounting FS... mounted!
> reading config file
> opened config file
> 
> parsed json
> applying offsets
> parsed config:
> {"Name":"iSpindel","Token":"","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":196.43,"SSID":" ","PSK":","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":73,"aY":-781,"aZ":883}
> woken from deepsleep, normal mode
> Spl 0: 83.67
> Spl 1: 83.45
> Spl 2: 84.07
> Spl 3: 85.66
> Spl 4: 85.17
> Spl 5: 88.86
> Spl 6: 72.44
> 
> a: -5906	15844	-482	absTilt: 84.30	T: 28.07	V: 4.10	owT: 27.31	Gravity: 30.70
> After waiting 0.00 s, result 3
> 192.168.0.157
> UBIDOTS: posting
> POST /api/v1.6/devices/iSpindel?token=HTTP/1.1
> Host: things.ubidots.com
> User-Agent: iSpindel
> Connection: close
> Content-Type: application/json
> Content-Length: 65
> 
> {"tilt":84.30,"temperature":27.31,"battery":4.10,"gravity":30.70}HTTP/1.1 200 OK
> Server: nginx
> Date: Wed, 28 Jun 2017 17:40:21 GMT
> Content-Type: application/json
> Transfer-Encoding: chunked
> Connection: close
> Vary: Accept-Encoding
> Vary: Accept, Cookie
> Allow: POST, OPTIONS
> 
> 8d
> {"battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "gravity": [{"status_code": 201}]}
> 0
> 
> 
> Final-sleep: 15s; RT:3113
> rl[00]$œŸ|[00]Œlà|[03][0C][0C][0C]Œ[04]dä[04]c|‡ƒ[03]ä[1B]›{’bŒ[04]b„ûogßlnnœâì[0C]c[1C]x„dsl{dpûgà[18][03][04][04]ƒ[0C]l[0C]Œœ[04][04][0C]c[0C]gã<[03]dä[0C][0C]‡c„ûooî[00]lŒŽd`[03]˜[12][13]gn[04]$`[03][07][02]gsŽ››o[0C][0C]c[04]œ[07]l[0F]{’’g[04][04]c[0C]œ[0F]dœ[03][04][04][04][0C][0C][0C][0C]d`[03]ü‚gœ[03]
> FW 5.1.2
> 2.0.0(656edbf)
> Worker run!
> Boot-Mode: 5
> mounting FS... mounted!
> reading config file
> opened config file
> 
> parsed json
> applying offsets
> parsed config:
> {"Name":"iSpindel","Token":"","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":196.43,"SSID":"","PSK":"W","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":73,"aY":-781,"aZ":883}
> woken from deepsleep, normal mode
> Spl 0: 86.67
> Spl 1: 87.33
> Spl 2: 86.15
> Spl 3: 87.51
> Spl 4: 86.77
> Spl 5: 86.33
> Spl 6: 86.59
> 
> a: -998	16492	-86	absTilt: 86.68	T: 28.39	V: 4.10	owT: 27.25	Gravity: 31.90
> After waiting 0.00 s, result 3
> 192.168.0.157
> UBIDOTS: posting
> POST /api/v1.6/devices/iSpindel?token=HTTP/1.1
> Host: things.ubidots.com
> User-Agent: iSpindel
> Connection: close
> Content-Type: application/json
> Content-Length: 65
> 
> {"tilt":86.68,"temperature":27.25,"battery":4.10,"gravity":31.90}HTTP/1.1 200 OK
> Server: nginx
> Date: Wed, 28 Jun 2017 17:40:35 GMT
> Content-Type: application/json
> Transfer-Encoding: chunked
> Connection: close
> Vary: Accept-Encoding
> Vary: Accept, Cookie
> Allow: POST, OPTIONS
> 
> 8d
> {"battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "gravity": [{"status_code": 201}]}
> 0
> 
> 
> Final-sleep: 15s; RT:2420
> rl[00]lœß|[00]„là|[03][0C][0C][0C]„[04]dä[0C]c|‡ƒ[03]ä[1B]›r’c„[04]cŒûggždooœâä[0C]c[1C]p„$sd{dpûoà[18][03][0C][04]ƒ[04]l[0C]Œœ[04][04][04]c[0C]gã|[03]dä[04][0C]‡cŒónoç[00]lŒŽd`[03][1B][12]g'[04]d`[03][0F][02]gsÇ“›o[0C][0C]b[0C]œ[07]l[0F]{›’g[0C][04]c[04]œ[0E]dœ[03][04][04][04][04][0C][0C][0C]l`[03]üƒgœ[03]
> FW 5.1.2
> 2.0.0(656edbf)
> Worker run!
> Boot-Mode: 5
> mounting FS... mounted!
> reading config file
> opened config file
> 
> parsed json
> applying offsets
> parsed config:
> {"Name":"iSpindel","Token":"","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":196.43,"SSID":"n","PSK":"","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":73,"aY":-781,"aZ":883}
> woken from deepsleep, normal mode
> Spl 0: 81.64
> Spl 1: 81.62
> Spl 2: 81.59
> Spl 3: 81.83
> Spl 4: 82.06
> Spl 5: 81.67
> Spl 6: 82.40
> 
> a: -2284	16292	124	absTilt: 81.71	T: 28.68	V: 4.10	owT: 27.25	Gravity: 29.39
> After waiting 0.10 s, result 3
> 192.168.0.157
> UBIDOTS: posting
> POST /api/v1.6/devices/iSpindel?token=HTTP/1.1
> Host: things.ubidots.com
> User-Agent: iSpindel
> Connection: close
> Content-Type: application/json
> Content-Length: 65
> 
> {"tilt":81.71,"temperature":27.25,"battery":4.10,"gravity":29.39}HTTP/1.1 200 OK
> Server: nginx
> Date: Wed, 28 Jun 2017 17:40:53 GMT
> Content-Type: application/json
> Transfer-Encoding: chunked
> Connection: close
> Vary: Accept-Encoding
> Vary: Accept, Cookie
> Allow: POST, OPTIONS
> 
> 8d
> {"battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "gravity": [{"status_code": 201}]}
> 0
> 
> 
> Final-sleep: 15s; RT:3242
> sl[00]lœž|[00]„là|[03][0C][0C][04]„[04]$ì[0C]c|Çƒ[03]ä[1B]Ûr“c„[04]cŒòggŸdooœãä[0C]#[1C]p„l{d{$xógà[10][03][04][0C]‚[0C]l[04]„œ[0C][0C][0C]c[04]nã|[03]lì[0C][0C]b„ûogï[00]d„$`[03]˜[13][13]oo[0C]l`[03][07][03]n{›Ûg[0C][04]c[04]œ[0E]l[07]r““n[04][0C]#[0C]Ü[0F]lœ[03][04][0C][0C][0C][0C][04][04]d`[03]üƒ'œ[03]
> FW 5.1.2
> 2.0.0(656edbf)
> Worker run!
> Boot-Mode: 5
> ...

padrino-xl, 2017-07-05:
Hi, 

did you have the time to check the readings I posted?

Right now I have a 350 ohm resistor installed. 
When the two pins are wired it seems to work. 
What's the smallest resitor I can use without risk? Or should I change to a diode?

Thanks for all your help and the great work here!

universam1, 2017-07-05:
The only danger is while flashing that the UART and the ESP short circuit. The resistor limits the current and a diode prevents that happening. I would not go under 200, a Schottky though might be best.

jvels, 2017-08-08:
Hi

I have the same issue, with the ispindle, only 1 upload and then nothings happens :-/ 

Would it help if remove my 470 ohm resistor and replaced with the following diode: https://www.reichelt.de/BA-BAY-BB-Dioden/BAT-43/3/index.html?ACTION=3&LA=446&ARTICLE=4851&GROUPID=2988&artnr=BAT+43&SEARCH=BAT43%2BSchottky%2BDiode  

Or are i'm misunderstand the conclusions from above?

universam1, 2017-08-08:
Post your terminal output here.

jvels, 2017-08-09:
Hi 

First i go to config mode, and setup, and then I get one result and then noting:

FW 5.2.1
2.0.0(656edbf)
Worker run!
Boot-Mode: 6
power-cycle or reset detected, config mode
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"velsdk","Token":"MY_TOKEN","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.800003,"SSID":"G12A","PSK":"MY_WIFI","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":429,"aY":-21,"aZ":3016}...........
ERROR no Wifi credentials

going to Config Mode

FW 5.2.1
2.0.0(656edbf)
Worker run!
Boot-Mode: 5

Double Reset detected
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"velsdk","Token":"MY_TOKEN","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.800003,"SSID":"G12A","PSK":"MY_WIFI","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":429,"aY":-21,"aZ":3016}...........
ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: G12A
*WM: -79
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: velsdk001
*WM: Parameter
*WM: sleep
*WM: 15
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 0
*WM: Parameter
*WM: token
*WM: MY_TOKEN
*WM: Parameter
*WM: server
*WM: 
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting wifi with new parameters...
*WM: previous settings invalidated
*WM: After waiting...
*WM: 6.00
*WM: seconds
*WM: Connection result: 
*WM: WL_CONNECTED
saving config...{"Name":"velsdk001","Token":"MY_TOKEN","Sleep":15,"Server":"","API":0,"Port":80,"URL":"","Vfact":191.800003,"SSID":"G12A","PSK":"MY_WIFI","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":429,"aY":-21,"aZ":3016}saved successfully
Spl 0: 63.99
Spl 1: 63.96
Spl 2: 63.92
Spl 3: 63.92
Spl 4: 63.95
Spl 5: 63.90
Spl 6: 63.96

a: 2008	1714	-6	absTilt: 63.94	T: 32.75	V: 4.00	owT: 29.87	Gravity: 20.30
After waiting 0.00 s, result 3
192.168.1.44
UBIDOTS: posting
POST /api/v1.6/devices/velsdk001?token=MY_TOKEN HTTP/1.1
Host: things.ubidots.com
User-Agent: velsdk001
Connection: close
Content-Type: application/json
Content-Length: 77

{"tilt":63.94392,"temperature":29.875,"battery":4.004171,"gravity":20.295227}HTTP/1.1 200 OK
Server: nginx
Date: Wed, 09 Aug 2017 14:09:19 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
Vary: Cookie
Allow: POST, OPTIONS

8d
{"battery": [{"status_code": 201}], "tilt": [{"status_code": 201}], "temperature": [{"status_code": 201}], "gravity": [{"status_code": 201}]}
0

Final-sleep: 1s; RT:119154

bralexc, 2017-08-09:
Yes, using a BAT43 Schottky diode instead of a 470 Ohm resistor should
solve your problem.

I used a BAT85 instead (didn't have a BAT43 around) and it's working
perfectly.

html?ACTION=3&LA=446&ARTICLE=4851&GROUPID=2988&artnr=BAT+
43&SEARCH=BAT43%2BSchottky%2BDiode

universam1, 2017-08-09:
So the last line is important,is there no further output?
If really so, then you have actually no connection between D0 and RST, not even over resistor! Well in this case the diode would help obviously 😉

jvels, 2017-08-09:
Hi

I use this circuit: [https://pcbs.io/share/8gbKR](https://pcbs.io/share/8gbKR )

[image]
[image]

So there should be a resistor?

/Jesper

universam1, 2017-08-16:
solved

closed 2017-08-16 by universam1

fogarasia, 2019-07-11:
> I used a BAT46 instead of a BAT43 and was able to get readings until 3.67v then it stopped communicating. Looks like a 197.36hrs between diode and using a BAT46 currently recharging the battery and rerunning the test again.

Same issue here too. Below 3,67v stops communicating using BAT46 (as only this was available at the local store), not sending data to Ubidots and not getting back from the deep sleep.

JETET, 2019-07-17:
This have sold me
[image]

JETET, 2019-07-17:
What position does the BAT place in?

fogarasia, 2019-07-17:
Possibly it was a fault Wemos or something with the PCB board. Anyhow, I have made a new one, with the same components, but new Wemos and now is working. But thanks!

JETET, 2019-07-17:
Have you installed the BAT43?

fogarasia, 2019-07-17:
No, because I Couldn't find one nearby, so I had only BAT46, which is working well. Now I am on the testing phase. The thing to resolve is to fix the sled to not slip deeper in the Petling, to hold the right tilt angle.

JETET, 2019-07-17:
I'm going to try the BAT 43 that I bought here very expensive € 1.95

fogarasia, 2019-07-17:
That was really expensive. Order from Ebay or Ali, cheaper but needs 40 days.

JETET, 2019-07-17:
Do you know what position it is in? He gave BAT

fogarasia, 2019-07-17:
Poiting to D0

JETET, 2019-07-17:
The tasting looking towards D0?

fogarasia, 2019-07-17:
Yes, the simbol for the cathode https://images.app.goo.gl/KeJWW6c37EyUiDHw6

[24 more replies]