universam1/iSpindel #61: DS18B20P vs Ds18b20

pronkster, 2017-05-02:
I have been troubleshooting my temperature reading as it was 5 degrees too warm and after replacing the temperature sensor a few times (three times, one from another supplier plus I have re-soldered the whole board) I figured out that the sensors is a parasitic power only version...  Is there any chance you could add support for this version in your code? 

Thanks

universam1, 2017-05-02:
I assume it is already working, since parasitic power I've tested as well successfully. The 5 deg. off is usually only related to heat transmission from the ESP during config time or if you run short intervals like 15sec. During normal operation with update interval >10min you wont have this issue.

pronkster, 2017-05-02:
Right, it turns out that the DS18B20P only requires 2 wires to connect. After I disconnected pin 3 from VCC I am getting the correct value . (as per : https://datasheets.maximintegrated.com/en/ds/DS18B20-PAR.pdf)

I'll see if I can put something in a document regarding this over the weekend and upload it to your site.

Thanks for your help! 👍

pronkster, 2017-05-02:
sorry, removed previous comment as it wasn't true ( I thought it was fixed) 

It turns out the DS18B20P only requires 2 wires to connect; pin 1 and pin 2. I have removed pin 3 but I still get a 5 degree difference. I will wait a while as you suggested, see if that helps.

pronkster, 2017-05-02:
Looks like its working now! I left it for 30 mins and the temperature slowly came back down to normal. I cant believe how sensitive it is though, the moment I go into the setup, the temperature goes up by 5 degrees. 

I am not sure if its fixed it by disconnecting pin 3 from VCC or, if I just needed to leave it for a while in sleep mode like you suggested. I might reconnect pin 3 as a test later this week. 

Thanks for the help 👍

pronkster, 2017-05-17:
Both DS18B20P and DS18B20 work perfectly fine. The temperature rises quickly if you set the time too low (say 20 sec) so that's where I went wrong. Putting the timer at 10 Mins solved my issue. Thanks 👍

closed 2017-05-17 by pronkster