universam1/iSpindel #62: Question Vfact for battery voltage?

Torfriishansen, 2017-05-16:
Hi,

Thanks for a really great project. 

I do not get how the vfact variable is used.

Does analogRead(A0); not give a value from 0 to 1023, referenced to 3.2 V?

float getBattery() {
  analogRead(A0); // drop first read
  return analogRead(A0) / my_vfact;   Could you explain how this work?
 //return analogRead(A0) * (3.2 / 1023.0) * divider ratio(5.5);  Is it not this?
}

The divider ratio is based on the diagram with the extra 230 k Ohm from Vcc to A0.

Br

universam1, 2017-05-17:
If you take a look at the Wemos D1 mini schematics you will see that the A0 is already a resistor divider to 1.0V of the ADC input.
Hence, the 230k resistor to Vcc is the extension up to 5V.

With the ADC range to 1023 the divider therefore converts to actual voltage. Is is basically a rule of three to derive. 
Now if you need to fine tune your setup you can again use the rule of three:

```
new factor = current factor * measured voltage / displayed voltage
```
Might be reciprokal though....

universam1, 2017-08-16:
solved

closed 2017-08-16 by universam1