universam1/iSpindel #63: Add infill parameters for printing drawer

christoferjh, 2017-05-20:
Unsure what infill I should use, or if should be solid?

lekrom, 2017-05-26:
Should not matter much as it caries very little load. Mine came out great @ 15% (PLA) but you can probably go lower as long as you do not force it to hard into the preform and you are not too agressive when tightening cable ties and such.

universam1, 2017-05-29:
This is indeed not critical, I have good results with ABS, PLA, with 1mm perimeter and 15% infill, but no need to force any of these.

hocomadvies, 2017-08-01:
Sorry for asking but since I can not print 3d myself I have to look on https://www.3dhubs.com/ which file should I use to get the right one printed for the petling xl ? Or are all these files the same?

universam1, 2017-08-16:
depends on the size of your petling, the sizes are described.

closed 2017-08-16 by universam1