universam1/iSpindel #68: Gravity points after dot

Philusha1983, 2017-06-21:
Is it possible to get more than 2 digits after dot?
There is a big difference between 1.06 and 1.07 in gravity calculation.

universam1, 2017-06-22:
Good point, I do some major rework, could you please try this version?

[firmware.zip](https://github.com/universam1/iSpindel/files/1093955/firmware.zip)

Philusha1983, 2017-06-22:
I'll test it later today. 
By the way what platform do you create the BIN file? 
My brewing partner is a developer, So he asked for this info, maybe he can help with this project.

universam1, 2017-06-22:
I'm using platform.io

tomcohen, 2017-06-22:
@universam1 
hi.  i just add the platform.io plugin into vscode IDE. init the project (wemos board and arduino libraries). how do i proceed  from here to export this project to binary file? 
thanks.

universam1, 2017-07-20:
no answer, so closing

closed 2017-07-20 by universam1