universam1/iSpindel #69: wemos d1 mini pro FS problem

gimiki, 2017-06-22:
I have some problem with the firmware. I have a wemos d1 mini pro but every time i get this error

`FW 5.1.2
                                                                                                        2.0.0(656edbf)
                                                                                                                      Worker run!
                                                                                                                                 Boot-Mode: 6
                                                                                                                                             power-cycle or reset detected, config mode
                                                                                                                                                                                       mounting FS... ERROR: failed to mount FS!

                                                                                                                                                                                                                                ERROR config corrupted
                                                                                                                                                                                                                                                      ...........
                                                                                                                                                                                                                                                                 ERROR no Wifi credentials

           going to Config Mode
                               rll��|�l�|�l�b|����r�b�b��nn�lnn���bp��lrlrlp�n��l��bn�|l��b��nn�l��l`�nnl`nr���nb�lr��nb�l�l`�n�
                                                                                                                                FW 5.1.2
                                                                                                                                        2.0.0(656edbf)
                                                                                                                                                      Worker run!
                                                                                                                                                                 Boot-Mode: 5

                                                                                                                                                                             Double Reset detected
                                                                                                                                                                                                  mounting FS... ERROR: failed to mount FS!

                                                                                                                                                                                                                                           ERROR config corrupted
                                                                                                                                                                                                                                                                 ...........
                                                                                                                                                                                                                                                                            ERROR no Wifi credentials

                      going to Config Mode
                                          *WM: Scan done
`

I also tried with older version but same thing happen. I flash it with -fs 4m and 16m. Same issue.

How can i do? Thanks!

universam1, 2017-06-22:
I wonder if this is related to the Pro which was never tested... The d1 mini is the supported one. Maybe you could try to build an Arduino project that just formats the SPIFFS for you and after success flash the firmware.

pronkster, 2017-06-22:
I use the Wemos D1 mini Pro without any issues. I prefer it over the mini because of the external antenna option which significantly improves WiFi signal strength  

Try reformatting the file system and install again after maybe. I found the bellow script by a quick google ..

#include "FS.h"

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Mounting FS...");

  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }

  SPIFFS.format();
  Serial.println("Done.");
}

void loop() { }

closed 2017-08-16 by universam1

cwilling, 2017-10-05:
I know this issue is closed, but in case other are searching for solution to this problem, I also had it with a D1 mini Pro (2 of them actually had the same symptoms). I tried the sketch from @pronkster and it worked - thanks for that.

Interestingly, after then configuring the iSpindel, it reported:

    *WM: Sent wifi save page
    *WM: Connecting to new AP
    *WM: Connecting wifi with new parameters...
    *WM: previous settings invalidated
    *WM: After waiting...
    *WM: 6.36
    *WM: seconds
    *WM: Connection result: 
    *WM: WL_CONNECTED
    saving config...
    need to format SPIFFS: 1

i.e. the wemos did its own formatting as well.

Anyway, they're now both working as advertised.