universam1/iSpindel #70: No WiFi Credentials

Alchomist, 2017-06-22:
Can anyone shed any light on whats happening here please.
I've tried every possible combination of service type & tried to connect to different networks with correct lowercase passwords. Also swapped the 470 resistor for the BAT43, but the error msg below remains the same. 
I found it odd that the Wifi BasicClient sketch connected to the network just fine
Any insight would be much appreciated

I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: iSpindel000
*WM: Parameter
*WM: sleep
*WM: 10
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 2
*WM: Parameter
*WM: token
*WM: 
*WM: Parameter
*WM: server
*WM: 10.0.0.15
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: POLYN
*WM: -0.00031*tilt^2+0.557*tilt-14.054
*WM: Sent wifi save page
I2C ERROR: 0 con:0
ERROR: OW DISCONNECTED

Exception (9):
epc1=0x401049a5 epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys 
sp: 3ffffd90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff30:  086ff210 3ffefef0 3fff0704 60000600  
3fffff40:  4022edbe 3ffeaab4 3fff0704 402081bb  
3fffff50:  4022eec0 3ffeaab4 3ffefef0 08707e7d  
3fffff60:  401048a2 02a303b5 3ffefef0 60000600  
3fffff70:  40244945 3ffeaab4 3ffefef0 086ff210  
3fffff80:  4024498a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeff10 3ffead40 00000004 402445bf  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vb6958986
~ld

FW 5.1.2
2.0.0(656edbf)
Worker run!
Boot-Mode: 2
mounting FS... mounted!

ERROR config corrupted
...........
ERROR no Wifi credentials

going to Config Mode
rl d⸮⸮| ⸮l⸮|⸮l⸮b|⸮⸮⸮⸮{⸮cl⸮c⸮⸮gg⸮lnn⸮⸮⸮cp⸮⸮lsl{d⸮r⸮g⸮#⸮nbp ⸮

universam1, 2017-06-22:
That's a clear one, the ESP crashes because your DS18b20 is not found, hence the program cant continue/work.

Check the connection, resistor 4k7 all around the temp sensor until the you dont get the ` ERROR: OW DISCONNECTED` any more.

Alchomist, 2017-06-22:
Ahhhh thanks
Now I know where to look....... at the beginning

Alchomist, 2017-07-01:
Totally new build still wont connect.
Any idea what this code points to?

Final-sleep: 15s; RT:8368
sd l⸮⸮| ⸮d⸮|⸮l⸮c|⸮⸮⸮⸮{⸮b⸮b⸮⸮og⸮lno⸮⸮⸮cp⸮⸮$sd{dp⸮o⸮⸮$⸮⸮cg⸮|d⸮⸮c⸮⸮oo⸮ l⸮⸮d`⸮nol`n{⸮⸮⸮gc⸮ds⸮⸮oc⸮l⸮b⸮ld⸮$⸮⸮$⸮⸮o⸮
FW 5.1.2
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindel001","Token":"i52NINXZ2D6xkPVpx1wbgCx9kn","Sleep":15,"Server":"5000","API":0,"Port":80,"URL":"","Vfact":191.800003,"SSID":"NETGEAR56_EXT","PSK":"basiccomet979 ","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 81.21
Spl 1: 81.24
Spl 2: 81.25
Spl 3: 81.27
Spl 4: 81.28
Spl 5: 81.27
Spl 6: 81.28

a: -2002	17218	1848	absTilt: 81.26	T: 14.42	V: 3.31	owT: 11.63	Gravity: 29.16
After waiting 6.09 s, result 6
Rescue Wifi credentials
failed to connect

Final-sleep: 15s; RT:8364

universam1, 2017-07-03:
`WL_DISCONNECTED     = 6` means that the authetication against your Wifi was rejected. 
First charge your battery.
Try a different network, no special characters in the PSK.

Alchomist, 2017-07-04:
Thanks universam1, ubidots is up and running!

It appears the problem was trying to connect to the router through a wifi extender.
The brewery is 100m from the router so Raspberry pi & ipad connects through the wifi extender.
But iSpindle could only see the extender & not the router, so could not log-in.

Is there a way of logging in to the router via the extender when iSpindle can't see it?

universam1, 2017-07-04:
Actually wifi extender usually help a lot to gain more resilient connects....
What channel is your extender on, above 12? Or even 5ghz?

Alchomist, 2017-07-04:
2.4ghz
Channel 6
It seems to drop off the network as it moves away from the router. Even as it moves past 2 extenders. The Rpi's in the brewery seem to log-in ok

Alchomist, 2017-07-10:
Success!!
Reserving the IP address & rebooting both the router & extension did the trick.

Thanks for all your help

closed 2017-07-11 by Alchomist

concex17, 2020-09-17:
hi,
i have a similar problem, after flashing the fw, i see the ispindel network but i can't connect. from the iphone i can't even see the network.
below the log.
any help is appreciated, thanks.

 FW 6.3.1
2.2.2-dev(38a443e)
Worker run!
mounting FS... mounted!
ERROR: failed to load json config

ERROR config corrupted
scanning for OW device on pin: 5
No devices found!
scanning for OW device on pin: 12
Found device with ROM = 28 54 57 58 10 20 2 DB
  Chip = DS18B20
  Data = 1 74 1 4B 46 7F FF C 10 55  CRC=55
  Temperature = 23.25 Celsius,
offsets not available
Boot-Mode: Deep-Sleep Wake
...........
           ERROR no Wifi credentials

going to Config Mode
*WM: Scan done
*WM: No networks found
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: tempscale
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: uri
*WM: Adding parameter
*WM: db
*WM: Adding parameter
*WM: username
*WM: Adding parameter
*WM: password
*WM: Adding parameter
*WM: job
*WM: Adding parameter
*WM: instance
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: POLYN
started Portal
*WM: SET AP
*WM:
*WM: Configuring access point...
*WM: iSpindel
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started