universam1/iSpindel #73: Update Parts =>Sourcing documentation

dsvensson, 2017-07-11:
This url has been out of stock for a long time now, and they don't reply to mail, at least not english mail it seems, http://cachers-world.de/de/Petling-XL

Perhaps you could throw up links to some other blessed cases?

universam1, 2017-07-20:
This one here and he offers fitting sleds as well
https://3d-mechatronics.de/de/petling-3dm1-plastik-mit-deckel_239.html

closed 2017-08-16 by universam1

dsvensson, 2017-08-16:
And it's this one that fits or is the name misleading?

https://github.com/universam1/iSpindel/blob/master/drawer/iSpindel%20mechatronics.stl

universam1, 2017-08-16:
exactly, and you can order it there as well

bschuhma, 2017-08-16:
Has anyone figured out how to get one shipped to the US?  3d-mechatronics.de only ships to Europe it seems. :-(

Thanks,

Bret

Philusha1983, 2017-08-16:
> protection.outlook.com/?url=https%3A%2F%2Fgithub.com%
> 2Funiversam1%2FiSpindel%2Fissues%2F73%23issuecomment-
> 322717552&data=02%7C01%7C%7C8e6247d85b8e48ef4d8b08d4e489d10e%
> 7Cfa7b1b5a7b34438794aed2c178decee1%7C0%7C0%7C636384727941959133&sdata=
> auth%2FAClOmH6WJRAvV_ON4qh78WkwKi8L2LLRks5sYrdXgaJp
> ZM4OU58L&data=02%7C01%7C%7C8e6247d85b8e48ef4d8b08d4e489d10e%

you can find in this store on eBay:
https://www.ebay.com/sch/uniquegeocaches/m.html?_nkw=&_armrs=1&_ipg=&_from=

5 pcs XXL set of BIG PET Micro Geocaching container
 you can buy then as single aswell.
the size is similar but you will need to do some sand paper work on the 3d
printed part.
you can order the XL version but they smaller and you will need to do
changes in the 3d printed part to fit the smaller part.

Philip   ;)

bschuhma, 2017-08-16:
Thanks, Philip.  I was actually hoping to get all the parts like the 47 Euro DIY kit at 3d-mechatronics.de so I don’t have to source each individually and/or MacGuyver it fit.

Regards,

Bret

https://www.ebay.com/sch/uniquegeocaches/m.html?_nkw=&_armrs=1&_ipg=&_from=

Philusha1983, 2017-08-16:
> armrs=1&_ipg=&_from=
> 322848722&data=02%7C01%7C%7Cb80e07de508e44cc70f808d4e4cf597b%
> 7Cfa7b1b5a7b34438794aed2c178decee1%7C0%7C0%7C636385026592918536&sdata=
> yYHkxBHxZ25xBZ28V_E9Y9ks5sYyv_gaJpZM4OU58L&data=02%7C01%7C%
> 7Cb80e07de508e44cc70f808d4e4cf597b%7Cfa7b1b5a7b34438794aed2c178de
> cee1%7C0%7C0%7C636385026592918536&sdata=Nrm%2BBx%

I can send you parts list if you want
 

*Philip* :)

bschuhma, 2017-08-16:
Thanks for the kind offer, Philip.  I believe I’ve seen the parts list at the project site on github, but going to 3-5 different sources and paying shipping at each will about double my costs.  Maybe I’ll approach the people at 3d-mechatronics.de and see if they’re amenable.

Thanks again! :-)

Regards,

Bret

Philusha1983, 2017-08-16:
> 322851189&data=02%7C01%7C%7C8bd7e598e5de4639017908d4e4d098f6%
> 7Cfa7b1b5a7b34438794aed2c178decee1%7C0%7C0%7C636385031951703846&sdata=WSO%
> 2FAClOmNFKHYGpQsfgWBq1kZGM3SjpAIKEks5sYy4OgaJpZM4OU58L&data=02%7C01%7C%
> 7C8bd7e598e5de4639017908d4e4d098f6%7Cfa7b1b5a7b34438794aed2c178de
> cee1%7C0%7C0%7C636385031951703846&sdata=kGlrAwfQo4LvwZULJHDM0Y5%

I ordered most of the parts from aliexpres and it all were less than 45$ 
per piece

*Philip* :)

bschuhma, 2017-08-16:
Did you send the sled info to a 3D online printing service?  I checked about 6 months ago and I seem to recall it was $27+shipping.  AliExpress isn’t bad unless you’re in a hurry! ;-)  Cheap parts, but sloooooowwww  ;-).

Thanks,

Bret

bschuhma, 2017-08-16:
I’ll take another look at the parts and sourcing.  If I can do it for under $70USD it’ll be worth it – that’s what the kit would cost if I lived in Germany.  I’m sure my shipping will be higher, though, so maybe my breakeven point is closer to $90USD.

Regards,

Bret

Philusha1983, 2017-08-16:
> 322852777&data=02%7C01%7C%7C5920df6d449f485ccaff08d4e4d16356%
> 7Cfa7b1b5a7b34438794aed2c178decee1%7C0%7C0%7C636385035352966522&sdata=
> 2FAClOmDatvMtJ4CeMtMmi4h6dhdszn0SAks5sYy9rgaJpZM4OU58L&data=02%7C01%7C%
> 7C5920df6d449f485ccaff08d4e4d16356%7Cfa7b1b5a7b34438794aed2c178de
> cee1%7C0%7C0%7C636385035352966522&sdata=EcfNg2jSZHtZjQQeJAKkqPDir0fiNm

In my country they opened new post sorting warehouse so AliExpress is
arriving in 10-14 days.
I printed the 3d part locally at a 3d printing service less than 20$