universam1/iSpindel #75: Battery voltage in ubidots always stays the same

wvdbroek, 2017-07-19:
Hi, I used my spindel for the first time in a brew last two weeks.
Now my voltage indicator in ubidots stayed all the two weeks at 4,45V when i measured it at the end of fermentation it was 3,70 V. What can i do to get ubidots to read the right voltage ?

Regards Willem

universam1, 2017-07-19:
Pretty sure your 220k resistor is not in place and the analog input is just saturated

wvdbroek, 2017-07-19:
where can i see this ? i did not make the spindel myself and bought it from somebody on a brewforum 
And can i easily lotr in a resistor or is this a no no on a finished one !!

universam1, 2017-07-20:
Actually, the license forbids to sell the iSpindel by third party, so this is not legal! Who is offering this?

wvdbroek, 2017-07-20:
this is a friend who made this for me because iam not good at soldering. So nobody is offering anything commercial. sorry for the mis understanding !! i had contact with him about this problem and he will have a look for me !!

wvdbroek, 2017-07-21:
problem solved turns out that he accidently put in a 220 resistor in stead of 220k. He now soldered in a 220k one and everything works fine !!

closed 2017-07-21 by wvdbroek