universam1/iSpindel #76: Generic TCP: Additional Field "UserToken"?

DottoreTozzi, 2017-07-20:
As mentioned here:
http://hobbybrauer.de/forum/viewtopic.php?p=224288#p224288

For the public server we are developing, an extra input field in the GenericTCP config section, representing a User Token in analogy to Ubidots would be nice-to-have.

The contents should be submitted within the JSON as an additional parameter in order to ensure backwards compatibility.

universam1, 2017-07-20:
Sounds good to me!
Would it make sense to hard code the DNS already for better user convenience?

DottoreTozzi, 2017-07-20:
ispindle.de and Port 9501 as defaults would be cool.
Later on I'm planning to semi-autoconfig the iSpindel by a URL shown by the server.
Good thing you insisted on keeping HTTP GET. :)

DottoreTozzi, 2017-07-20:
We'd need the IP to remain configurable, though, for those who want to use a Raspi as a Relay, of course...

universam1, 2017-07-20:
K cool, I am also thinking about implementing a conditional config fetch, lets say every n-hours. This won't cost as much battery lifetime as on every upload cycle and still provide remote configuration.

DottoreTozzi, 2017-07-20:
I absolutely love this idea! That would be brilliant!
It could be triggered by a special return character from the script, within the same TCP connection.
So, once every n timer intervals the iSpindel would actually listen before shutting down the connection.

ckrack, 2017-07-20:
+1 for the token and yes, please keep the URL editable.
It'd be possible that users host this on their own server.

ckrack, 2017-07-20:
For backwards compatibility on older firmware we can allow the token to be sent in the name field, too.

DottoreTozzi, 2017-07-20:
Let's not touch the name field. People want to use it, especially in connection with Ubidots.
A separate field is perfectly backwards compatible with all versions of the local server, and if implemented the way Sam suggested (nothing sent if empty), would remain so even with any other homebrew (pun intended) solution.

ckrack, 2017-07-20:
Ok. Is there any kind of state of the art method of authorization in TCP APIs?
Somethin like the Authorization Header in HTTP for Restul APIs..
If so, we should use this.

Don't get me wrong, it's just a for fun and hobby thing but IOT got the bad press about security issues already, so let's not add to this ;)

DottoreTozzi, 2017-07-20:
Like I said before, there is no such thing as a TCP API unless it's being implemented on Application Level.
Let's discuss this via PN on the forum and not unneccessarily contribute to Sam's workload by bloating this thread here.
You'll receive a detailed reply from me within the hour. :)

universam1, 2017-07-20:
If it comes to security, then the real step is to use SSL which again becomes really expensive in battery lifetime. 
And as speaking about IoT without checking the key chain there is also very limit benefit of doing this. 

Conclusion, if rock solid security is not achievable, why bother too much.

DottoreTozzi, 2017-07-21:
Fully agreed. There's no sensitive data to be gathered, here, and the iSpindel can't be hijacked to become part of a bot net, since it only speaks and doesn't listen. Server side, I'll implement some counter measures against flooding and brute force. Definitely won't need SSL for fermentation data ;)

universam1, 2017-08-13:
Bumping on to do list

universam1, 2017-08-14:
[firmware.zip](https://github.com/universam1/iSpindel/files/1222441/firmware.zip)

Can someone @DottoreTozzi @ckrack please try and give feedback? Thanks!

jvels, 2017-08-21:
Hi

I have tried firmware, but it end with config portal:
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: G12A
*WM: -76
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: name
*WM: velsdk002
*WM: Parameter
*WM: sleep
*WM: 15
*WM: Parameter
*WM: vfact
*WM: 191.80
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: 
*WM: 
*WM: Parameter
*WM: selAPI
*WM: 3
*WM: Parameter
*WM: token
*WM: PzMPHVxsq
*WM: Parameter
*WM: server
*WM: MYDOMAIN.com
*WM: Parameter
*WM: port
*WM: 80
*WM: Parameter
*WM: url
*WM: /beer/
[... 126 lines cut ...]
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

FW 5.3.0
2.0.0(656edbf)
Worker run!
Boot-Mode: 2

Double Reset detected
mounting FS... mounted!
reading config file
opened config file

parsed json
applying offsets
parsed config:
{"Name":"velsdk002","Token":"PzMPHVxsq","Sleep":15,"Server":"MYDOMAIN.com","API":3,"Port":80,"URL":"/beer/","Vfact":191.8,"SSID":"G12A","PSK":"MY_WIFI","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1513,"aY":97,"aZ":1657}
going to Config Mode
{l

universam1, 2017-08-21:
thanks for testing! I unfortunately dont have the backend available to test against, so I wonder how to proceed here.

DottoreTozzi, 2017-08-21:
On it.

DottoreTozzi, 2017-08-22:
Entering the Token parameter on the iSpindel portal worked, the value doesn't seem to be transmitted in the JSON though, as of yet:

`waiting for connection... listening on port: 9501
('192.168.178.56', 7971) received:'{'
('192.168.178.56', 7971) Input Str is now:{
('192.168.178.56', 7971) received:'"name":"iSpindleTest01","ID":"263990","angle":85.6939,"temperature":27.5,"battery":4.285714,"gravity":31.40103}'
('192.168.178.56', 7971) Input Str is now:{"name":"iSpindleTest01","ID":"263990","angle":85.6939,"temperature":27.5,"battery":4.285714,"gravity":31.40103}
('192.168.178.56', 7971) iSpindleTest01 (ID:263990) : Data received OK.
('192.168.178.56', 7971) - closed connection
(`

universam1, 2017-08-25:
which provider did you select? Do you have the terminal output of the ispindel handy?

DottoreTozzi, 2017-08-25:
GenericTCP.

Just tried again, this is what the serial monitor says:

parsed json
offsets not available
parsed config:
{"Name":"iSpindleTest01","Token":"3f4897a52fa10562","Sleep":900,"Server":"192.168.178.43","API":6,"Port":9501,"URL":"","Vfact":191.8,"SSID":"TOZZINET","PSK":"xxxxxxxxxxxxxxxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 87.04
Spl 1: 87.07
Spl 2: 87.06
Spl 3: 87.15
Spl 4: 87.08
Spl 5: 87.06
Spl 6: 87.04

a: 302	15428	740	absTilt: 87.06	T: 32.04	V: 4.33	owT: 29.62	Gravity: 32.09
After waiting 1.60 s, result 3
192.168.178.56

calling DTTCP 
genericTCP: sending
server: 192.168.178.43 Port: 9501
{"name":"iSpindleTest01","ID":"263990","angle":87.06411,"temperature":29.625,"battery":4.327425,"gravity":32.09085}wait

Final-sleep: 900s; RT:3455

universam1, 2017-08-28:
Thanks @DottoreTozzi @jvels for testing - in response I've completelly rewritten the upload process to have it more granular and flexible. From my testing the token is now successfully added to the json string.
Please check this out and report back.

universam1, 2017-08-28:
Here @DottoreTozzi @jvels is the new compiled firmware. Thanks for testing!
[firmware.zip](https://github.com/universam1/iSpindel/files/1258363/firmware.zip)

DottoreTozzi, 2017-08-29:
Token is transmitted OK now, however the iSpindle's name is now missing from the JSON...

ckrack, 2017-08-29:
If you use your hosts file and re-route ispindel.de to 127.0.0.1, you can test it.

How do we go about the preference of the name fields?
The workflow to add a spindle to the public-server is to generate a token and set a name to be displayed in the interface.
The name that is generated from the spindle-ui is not needed.
I can imagine people using the public-server more often than the spindle-ui, as you have to go into config mode before dropping it into your fermenter.
This would argument for prefering the name set in the public-server and omitting the one set in the spindle-ui.

universam1, 2017-08-29:
@ckrack the iSpindel has no hosts file!

Regarding the name field, whatever you use to display in the web service is up to you, you can map to the `ID` or the `name` property, both are now transmitted.

DottoreTozzi, 2017-08-29:
OK one final small quirk:
ID used to be transmitted in double-quotes, now these are ommitted.
If I fix this on my side (server), the new firmware would break existing installations...

jvels, 2017-08-29:
@universam1 I can confirm it works :-)

universam1, 2017-08-29:
Thanks @jvels ! 👍

You @DottoreTozzi are actually correct, while working on it it came to my mind that the ID is an integer serial number I've been transmitting as a String which kind of was nonsense... So I've changed it now to its native data type. 
I'm not so much a fan of keeping the faults of the past just because of history. But if you really feel strongly about that matter let me know I will do you the favor 😉

DottoreTozzi, 2017-08-29:
Thank you Sam!

Well, I am kinda torn by this, on the one hand I'd really like to keep backwards compatibility, on the other hand I absolutely agree that keeping incoherencies from the past and work around them is not the way to go in the long run.

I do not know how many people are actually using the local server at the moment, but I guess it's in the lowest 2-digit range.

So, now will probably be the best time to set things right, rather than later.

I will incorporate a fix and warn people that they will have to update the script and database in order to be able to use the new firmware and will make sure I get the fix out ASAP.
Need to do a major review and some nasty GIT merges over the next few days anyway, as well as put together a new Raspi image and create a repository for the public server, so you won't be bothered by that anymore.

Thanks again for your work, really nice job, and I guess we can close this issue now.

universam1, 2017-08-30:
Awesome @DottoreTozzi great work! So I'll wait for the merge until you got the change in, so I can mention that required update in one go.
Looking forward!

DottoreTozzi, 2017-08-31:
Actually, you might as well merge it right away... :)
As it turns out, older versions of my script do work fine after all, it's just the debug output that's screwed up.
By a lucky mistake of mine the script would send an NAK (which the firmware doesn't listen to), but else behave just normally and write the data anyway. That's a bug I'd almost like to call a feature...
Python 2.7 is apparently way more tolerant when it comes to these things than I had expected.

But I'll submit a pull request anyway. Quite a few changes I'd like to release now.

DottoreTozzi, 2017-09-28:
Looks like in the latest version, iSpindel "Name" is no longer being transmitted.

universam1, 2017-09-28:
Well, it's in there https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L433 but lower case, might the be the issue?

I can't test right now as my iSpindel are currently busy producing beer ... 😄 

Do you have a link I can mention?

DottoreTozzi, 2017-09-28:
I assembled a new iSpindel last night and directly flashed 5.4.0 on it (firmware.zip from the link above, version 28.08.2017).
Token is transmitted OK, but "Name" field is missing.
Seems to be a separate issue though, as the version from August 16th worked fine.

The link to my local server repository is:
https://github.com/DottoreTozzi/iSpindel-TCP-Server

universam1, 2017-09-28:
Could you give 5.5 a try and post the json output from console? I’ll try to reproduce ASAP.

DottoreTozzi, 2017-09-28:
5.5 works. Sorry for the confusion...

closed 2017-09-28 by DottoreTozzi

universam1, 2017-09-28:
Cool thanks for testing