universam1/iSpindel #77: question for FG instead of Plato

panzo19, 2017-07-27:
Is there any plan to create FG.php instead of Plato.php or is there any optio to transform Plato to FG  in php file?

universam1, 2017-07-27:
what php file are you referring to?

panzo19, 2017-07-27:
in forlder tools/generic-TCP/web

universam1, 2017-07-27:
Oh ok that's contribution from @Tozzi, you may open an issue on his repo.

pumalyf, 2017-08-07:
I'm interested in turning into FG also, can you share any news?

universam1, 2017-08-07:
@pumalyf @panzo19 you want to open an issue at @DottoreTozzi repository!

DottoreTozzi, 2017-08-07:
I will look into it after my vacation.

universam1, 2017-08-14:
@DottoreTozzi since your contribution is becoming a project, would it make sense to split it out into a dedicated project?

jvels, 2017-08-19:
A input for @DottoreTozzi I have been playing with the chart lib. And have tried to make a "linked chart". It is "fake" data I have used. It might do plato/FG chart easier to read. Just an input.
[image]

universam1, 2017-10-10:
I linked the repo of dottoreTozzi in the Readme, feel free to continue development there.

closed 2017-10-10 by universam1