universam1/iSpindel #79: monitor.beer now support iSpindel

3sverd, 2017-08-11:
Not an issue, just to say that https://monitor.beer nor support iSpindel 100%, you can now show full fermentation result. No longer need for creating your own graphs etc, just registrer and enjoy.

Continuous alcohol%.
Continuous yeast attenuation%.
Continuous fermentation%.
Continuous temperature.
Ongoing SG.
How long sg is stable for days.
Minimum temperature through fermentation.
Average temperature through fermentation.
Maximum temperature through fermentation.
Number of days fermentation.

kabelbrand, 2017-08-11:
Naa, 3 month free, then it costs! Must write: Just register, pay and enjoy. Poor advertising...

universam1, 2017-11-17:
assume it is fine, so closing for now

closed 2017-11-17 by universam1