universam1/iSpindel #80: upload to http php exsample

jvels, 2017-08-13:
Hi

I'm not a github expert, so I post my contribute here. It is php script there save data in a mysql database (instead of using the python script). It use the same database table use by the TCP example. Just insert db server,  database name, user and password.

In the config of the i spindle set:
[image]

Save the below code as index.php and place it domain.com/beer/index.php and the it will recive the data from the ispindle and store it in the database.

`<?PHP

$inputJSON = file_get_contents('php://input');

$data = json_decode($inputJSON);

$date = date("Y-m-d H:i:s");
 
$servername = "XXXX";
$dbname = "XXXX";
$username = "XXXX";
$password = "XXXX";

// Create connection
$conn = new mysqli($servername, $username, $password, $dbname);
// Check connection
if ($conn->connect_error) {
    die("Connection failed: " . $conn->connect_error);
} 

$sql = "INSERT INTO Data (Timestamp, Name, ID, Angle, 	Temperature, Battery, ResetFlag, Gravity)
VALUES ('". $date. "', '". $data->name ."', '". $data->ID ."','". $data->angle ."','". $data->temperature ."','". $data->battery ."','','". $data->gravity ."')";

if ($conn->query($sql) === TRUE) {
    echo "New record created successfully";
} else {
    echo "Error: " . $sql . "<br>" . $conn->error;
}

$conn->close();

?>`

universam1, 2017-08-13:
Thanks for the snippet, will keep it around as an example!

closed 2017-08-30 by jvels

tonarii, 2018-10-08:
Hi, i have an issue trying to add the Interval and RSSI value into the insert into php file.
I have an error with "Interval" table.
I creatade the table on mysql.
Here is my table :
`CREATE TABLE `Data` (
	`Timestamp` datetime NOT NULL,
	`Name` varchar(64) COLLATE ascii_bin NOT NULL,
	`ID` INT UNSIGNED NOT NULL,
	`Angle` double NOT NULL,
	`Temperature` double NOT NULL,
	`Battery` double NOT NULL,
	`ResetFlag` boolean,
	`Gravity` double NOT NULL DEFAULT 0,
	`UserToken` varchar(64) COLLATE ascii_bin,
	`Interval` int,
	`RSSI` int,
	PRIMARY KEY (`Timestamp`, `Name`, `ID`)
	)
ENGINE=MyISAM DEFAULT CHARSET=utf8
COLLATE=utf8_general_ci AUTO_INCREMENT=1 COMMENT='iSpindle Data';`

And this is my insert into php code :
`$sql = "INSERT INTO Data (Timestamp, Name, ID, Angle, Temperature, Battery, ResetFlag, Gravity, UserToken, Interval, RSSI)

VALUES ('". $date. "', '". $data->name ."', '". $data->ID ."','". $data->angle ."','". $data->temperature ."','". $data->battery ."','','". $data->gravity ."','','','". $data->rssi ."')";
`
If you can help me to update this :)