universam1/iSpindel #83: Always 5.338895 voltage

jvels, 2017-08-16:
Hi

Any idea why my ispindle show 5.338895V ?

I have assambled one where voltage goes down after usage as it should, but this one still display 5.338895 v after 12 hours usage with submit every 15 sec?

jvels, 2017-08-16:
FW 5.2.1
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"velsdk003","Token":"","Sleep":15,"Server":"xxxxx","API":3,"Port":80,"URL":"/beer/","Vfact":191.800003,"SSID":"G12A","PSK":"xxxxxx","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 57.56
Spl 1: 57.61
Spl 2: 57.46
Spl 3: 57.45
Spl 4: 57.32
Spl 5: 57.78
Spl 6: 57.92

a: -5338	11440	-6548	absTilt: 57.54	T: 30.10	V: 5.34	owT: 28.00	Gravity: 16.97
After waiting 1.50 s, result 3
192.168.1.42

calling DTHTTP 
genericHTTP: posting
POST /beer/ HTTP/1.1
Host: vels.dk
User-Agent: velsdk003
Connection: close
Content-Type: application/json
Content-Length: 109

{"name":"velsdk003","ID":"3110168","angle":57.542564,"temperature":28,"battery":5.338895,"gravity":16.970753}HTTP/1.1 200 OK
Date: Wed, 16 Aug 2017 14:54:13 GMT
Server: Apache
X-Powered-By: PHP/5.6.30
Vary: Accept-Encoding
Connection: close
Transfer-Encoding: chunked
Content-Type: text/html; charset=UTF-8

1f
New record created successfully

universam1, 2017-08-16:
you got a shortcut from Vcc to A0 instead of using the 220k resistor

closed 2017-08-16 by jvels