universam1/iSpindel #88: Minimizing the 3d drawer for smaller PET

Philusha1983, 2017-08-31:
I want to create a smaller version of the drawer.
Are there technical issues that I need to take into consideration? 
[image]

bralexc, 2017-08-31:
You shouldn't have any issues other then the reduced turned on time due to
smaller battery capacity.

In fact, someone I know made one out of a Falcon tube using a 14500 battery
(2200 mAh).

deedos, 2017-10-18:
"In fact, someone I know made one out of a Falcon tube using a 14500 battery
(2200 mAh)." 
That' s cool. Does he have a project for the drawer ? Is it ok just replace de battery?

bralexc, 2017-10-26:
No project, sorry. He had to remove the TP4056 due to size restrictions, so to recharge he always needs to take the battery out.

universam1, 2017-11-17:
assume it is fine, so closing for now

closed 2017-11-17 by universam1