universam1/iSpindel #89: English circuit diagram is not in english

XREvo, 2017-08-31:
The image at [circuit_diagram_en](https://github.com/universam1/iSpindel/blob/master/docs/https://github.com/universam1/iSpindel/blob/master/docs/circuit_diagram_en.md.md) is not in english (since the page is related to english content).
Can you translate it?

Paiakan, 2017-10-07:
[image]
[image]

Done. I also included the BAT43 diode as option to the 470 ohm resistor and also included a Portuguese version of the diagram.

By the way, I can help translating the "READ ME" file to portuguese, if you are interested.

universam1, 2017-10-08:
Thank you @Paiakan this is highly appreciated! 👍 

I would love to include other languages as well and feel free to open a pull request for updates in this regard.

Paiakan, 2017-10-09:
Weel, I am new in this kind of collaboration. Tried to translate creating a fork but I believe my first atempt didn't work.

XREvo, 2017-10-11:
Thank you @Paiakan

universam1, 2017-10-11:
merged your contributions, thank you!

closed 2017-10-11 by universam1