universam1/iSpindel #92: It's asking for Token with HTTP

zandoga, 2017-09-05:
[image]

universam1, 2017-09-05:
What's the error?

zandoga, 2017-09-05:
Doesn't communicate with BrewPiLess, it was communicating with 5.2.3.. If
it's ok to leave Token empty than nevermind, please close the issue. I'll
try again.

universam1, 2017-09-05:
Exactly, if the token is left empty it is skipped. Please try and report.

zandoga, 2017-09-06:
Tried again, no luck. Reversed back to 5.2.3 and it also doesn't work.
Ubidots works just fine. Looking at the serial monitor it seems that it
connects to BrewPiLess but the gravity doesn't get displayed. It's probably
an user or BrewPiLess issue.

ksga, 2017-09-18:
Had the same issue... Try to select another comm option, and then reselecting http - now the options should show correctly...

closed 2017-10-10 by universam1