universam1/iSpindel #93: Issue connecting iSpindle to BrewPiLess

pronkster, 2017-09-08:
I am having issues connecting iSpindle to brewpiless. Everything was working fine before I installed firmware version 5.2.1. Ubidots is working fine and I can see network connectivity is up. I can ping when it is out of sleep.. The monitor is showing no errors. I have re-installed a few times but still no luck.. Any idea what I can do next?

Thanks

lekrom, 2017-09-14:
I have the same problem...

universam1, 2017-10-06:
Please post the console output when a connection to BrewPiLess is established so we can discover what's going wrong.

universam1, 2017-10-08:
Here is a bugfixed version, using a library now to perform the post. Please also monitor the execution time in this regard and let me know how it works.
[firmware.zip](https://github.com/universam1/iSpindel/files/1366844/firmware.zip)

universam1, 2017-10-08:
closing in favor of #103

closed 2017-10-08 by universam1