universam1/iSpindel #94: Voltage conversion factor error

ksga, 2017-09-12:
HI.
I've been trying to adjust my voltage conversion factor but seem to have stumbled upon a bug of some sorts.. might be cheap Chinese hardware - might be some sort of error in the code...

Anyhow - I decided to find my conversion factor using trial and error (partly because I'm to lazy to fire up the calculator...).
Slowly zeroed in on but then got a "fatal exception" error on saving the config. After removing power the configuration would be completely reset (node name, conversion factor, WiFi settings, and gravity conversion)...
I've been able to replicate the error a number of times - it seems that it happens when entering **a conversion factor of 186,xx** (any decimal does the trick). 185 and 187 works fine, regardless of decimals.

Luckily 187,00 is quite close so I'll survive :)

ksga, 2017-10-11:
Got my second iSpindel soldered - and it seems to some sort of hardware error on the wemos.. 
Second instance have no issues with 186,xx (or any other value I tested...)

Back to the soldering table :)

DominikAuras, 2017-11-13:
Same here. My steps: flashed firmware 5.6.1 (binary, with the ESP Flasher). Connect to WiFi, opened configuration page. Clicked save. No changes. Conversion factor 191,80. It's already there. Crash seen in serial monitor.

Second time, changed vf to 200. Same.

At boot, it states: ERROR config corrupted.

```*WM: Sent wifi save page
ERROR: OW DISCONNECTED

Exception (9):
epc1=0x40104a4d epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  00000056 00000001 4010417e 3ffeeb90
3fffff60:  4022a0c5 3ffef970 00000001 00000000

3fffff70:  4023e683 00000004 00000000 00000000
3fffff80:  3fff2ab4 4023e032 3fffdab0 00000000
3fffff90:  3fffdcc0 3ffeaaf0 00000000 40249163
3fffffa0:  3ffeaaf0 40000f49 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

universam1, 2017-11-13:
Check other issues, when DS is not connected the FW is in unpredictable state. Not supported, make sure the peripheral is in place

closed 2017-11-13 by universam1