universam1/iSpindel #95: Couple questions to experienced users

Philusha1983, 2017-09-14:
Couple questions to experienced ispendel users:
After using ispendel for fermentation cycle.
You take out the electronic parts to charge the battery and the PET goes for cleaning - do you calibrate the ispendel again after gathering the parts for another use?
Is it possible to calibrate the ispendel just once?
Can you share your methods of use?

universam1, 2017-09-14:
The iSpindel ist not intended to be disassembled at all, thats the idea of the charging board. The design is to open the  cap only to switch it on/off and to plug the charger! No battery removal or anything like this. 

Hence, the iSpindel is to be calibrated once for ever. Of course if you disassemble it, you will need to start from scratch.

Philusha1983, 2017-09-14:
I see, but how you can clean the small gaps between the cap and the threads?
If you have super active fermentation - you will have dry yeast covering 
the threads.

*Philip* :)

universam1, 2017-09-14:
Sure, you need to remove the burned yeast. I open the cap and spray a good amount of Isopropyl alcohol on the thread and cap and clean with a towel. 
That works good, no infections.

closed 2017-09-20 by universam1