universam1/iSpindel #96: iSpindle Remote Configuration

DottoreTozzi, 2017-09-19:
As the topic came up again on hobbybrauer.de today (and I'm sure quite a few people would love to see a feature like this):

We should discuss what a remote configuration of the iSpindle from a local or public server might look like.

I'll implement this in the existing "local", usually Raspi-based solution first, since the Python script will also be the foundation for the future public server and a spare Raspi as an access point is useful, if not required, anyway, when using stainless fermenters as is the case in my setup, where the iSpindel's WiFi range is rather limited.

In order to keep it as simple as possible on both sides, I suggest it looks something like this:
There is a user defined interval in the iSpindel's config, like, daily, or every n sleep intervals, after which the iSpindel asks the generic TCP server for a JSON config string, mapped by hardware ID, verified by User Token.

For example, the iSpindle opens a TCP connection and sends a JSON like this:
{ cmd: "qryCfg", ID: xxxxxxxxxxxxxx, token: "xyzxyzxyzxyz" }

The server replies with a JSON containing all relevant config values.
The iSpindle then terminates the connection, checks if there are any changes to its config (in which case it will probably reboot), else simply reopens it and transmits its data as usual.

Impact on battery life should be minimal in this scenario...

universam1, 2017-09-20:
Good idea to discuss this topic here as to have some documentation.

I was also considering various options how to remote configure the device. 
Using JSON as the protocol is definitely the way to go and offers flexibility for future additions and extensions. 
I'm not sure yet what would be the best way to determine the update interval, as this is uptime which is directly costly battery life. It might be also remote configurable but will lead to user confusion as it would be always delayed by one iteration. Perhaps a hard coded setting could be enough to start with, or something incremental being very frequent at the beginning (new batch, everyone is nervous) and less often over the time (days, or weeks) passing by.

Though, regarding the communication protocol I feel strongly to focus on well tested implementations like REST or MQTT. On the first glance, a bare TCP connection might be easy in case of the Raspi setup but its major flaw is is not being portable at all. 

Considering a **REST** request is a well proven standard and the main benefit is its simple debugging. Regarding the python setup it is also easy to implement AFAIK. Hence, this API can be easily ported or used also with other services than the Raspi with just changing the URL.

I like the implementation of this one: https://ubidots.com/docs/api/index.html#get-a-variable and would suggest leaning towards it.

Regarding **MQTT**,  very fitting at the first glance, but it's major problem is that it is in contrast to REST a _stateful_ protocol.
We have a problem with a stateful protocol since every sleep we basically do a full restart. The initialization of a MQTT subscription is expensive regarding the handshake, and as with every restart we have to start from scratch, it cant outweigh a REST connection.

universam1, 2017-09-20:
Ok cool that looks like a base to start with. Let's keep updating the progress then. I'll start implementation based on the Ubidots specification that makes it easy to develop against.

DottoreTozzi, 2017-09-20:
Sounds brilliant!

Owemeone, 2018-03-01:
Wouldn't it be better to have a configuration update notification within the response of submitting the data? That way, you don't have to poll the configuration, when there's no change.

universam1, 2018-03-01:
Possible in a custom server but for Ubidots.

Owemeone, 2018-03-01:
I was indeed thinking of a custom server, as I assume your new public service will not be powered by Ubidots. I think it would be a hassle to update and change your configuration through Ubidots.

universam1, 2018-03-01:
@DottoreTozzi would you be able to also create a control with your project and return a json like { interval: xx } as within the response?

DottoreTozzi, 2018-03-01:
Sure, that won't be a problem!

gmorfeus, 2018-03-09:
If someone is interested, I've developed this functionality in my iSpindel. I've added needed stuff (address + URL) for this purpose and all needed to call and download new config. ISpindel sends a POST request with name, chipId and current configuration. On server side, PHP script can "change" this config or send a new one or just send empty response. ISpindel knows to handle empty replay and also does basic checking of received JSON before saving to memory. I have also added a possibility to put iSpindel in config mode from remote (like pressing reset button) and that happens when server script returns only "config" in body.
The purpose of all this is:
1. I have added reed switch to iSpindel
2. added wireless charging,
=> so once I power on Ispindel, I can close it, calibrate and don't have to open it again :)

universam1, 2018-03-09:
Thanks, I have finished it and running latest tests

ckrack, 2018-04-05:
How does the implementation look?

Do we return the config from the endpoint that receives data?
Is there a specific endpoint for GETing config?
Or did you follow the Ubidots implementation, where you need subsequent requests (one to get a list of variables + one for each variable).

What happens with empty values?
In a service operated by someone else, i would not want to store my wifi details for example.

It would be good to have some kind of beta channel, where we can test new features in advance of a release.

universam1, 2018-04-06:
Check out the current work in progress branch https://github.com/universam1/iSpindel/tree/remoteConfig and give it a try! I attached a current bin that you can just try out.

For now there is support for Ubidots using the variable interval by reading it back. Some screenshots you can see here http://braumagazin.de/article/ispindel-die-idee/ under 'Remote Konfiguration' they should be self-explanatory: If the iSpindel detects a changed value than the saved one, it will update it's configuration.

I just finished a real brew the other day using this remote config and had great results! The battery power usage was not noticeable higher.

Nevertheless, I am still working on an algorithm to smarter and less frequently call for an update to save power. 

Feel free to report your findings here.
[firmware.zip](https://github.com/universam1/iSpindel/files/1885486/firmware.zip)

ckrack, 2018-08-20:
Right now this seems to be focussed on ubidots.
I'd love a platform independent solution, while I understand that you want to support ubidots.
However, ubidots' implementation would require n requests for n values.
I think it would be best to separate this, as with the different options of endpoints.

There should be a GET to the specified URI.
The implementation of the endpoint should reply with a JSON dictionary containing all values that should be set in the config.

The basic flow:
ISpindel requests config data
`GET $ENDPOINT`

Endpoint returns JSON with HTTP Code 200
`{
key: data,
key2: data2,
...
}`

Why not just use the dictionary that is already in use? https://github.com/universam1/iSpindel/blob/master/pio/src/iSpindel.cpp#L162

It should be possible to omit certain values, as for hosted services i am quite sure that no one wants to put their WIFI credentials there, for example.

universam1, 2018-08-20:
There are basically two options to gain the configuration which would be `explicit` request against an endpoint to get the configuration on demand, or the `implicit` possibility by replying with a configuration dictionary from the server as an ack like @DottoreTozzi suggests.

For some backend we have no other option than the `explicit` request, like I demonstrate with Ubidots, simply because we depend on existing APIs. Thats basically what @ckrack you mention right.

It appears though that the `implicit` is way more power efficient because we get the configuration data delivered for free. This of course is possible only against a custom backend like @DottoreTozzi TCP server. Though thinking about twice we need to prove that first, since if might extend every single run in contrast to the `explicit` request which we can perform on demand. There will be a point where it pays off if we have a good algorithm to skip unnecessary request thus saving run time. I am not yet completely clear how it would look like, maybe a quadratic skip of interval based on changing settings.

What I would like to see is a comparison of the runtime (printed on console as `RT: xxxx`) with the payload of a JSON dict of `{ "interval": 600 }` versus the current `ack` only in order to determine the costs of the implicit data transmission. Could you @DottoreTozzi set this up?

DottoreTozzi, 2018-08-20:
Will do.

Perhaps we can improve timing in a further step, by flagging if there are changes.
For example, ACK for "OK, no config changes", ESC followed by JSON for "OK, but we do have config changes", and NAK for "sorry, come again?".

But let's first see what the impact is.

ckrack, 2018-08-20:
Please keep HTTP endpoints in mind.
In custom endpoints it would be possible to use `If-Modified-Since` headers and the timestamp of the config change - the device would only then run the logic to import config changes.
I think there are two things that need to be taken into account with the implicit way.
A) Is the request slower with a full json dictionary being returned (slower due to payload)?
B) Is it the import of the data which is having more impact?

We might be better off with a daily explicit call.
And we could also request an option for returning the latest values of a set of named variables for the ubidots api, which would be necessary to use the same implementation.

DottoreTozzi, 2018-08-20:
@universam1 check this out:
https://github.com/DottoreTozzi/iSpindel-TCP-Server/blob/send_cfg_json/iSpindle.py

The beauty of generic TCP is that this will probably generate no additional payload worth mentioning, as the reply as transmitted over the same stream. No GET requests, no overhead whatsoever.
Which is why the generic TCP solution currently has a payload of approximately 250 micro seconds (including debug output on the console), here, in my environment.
Which was exactly the initial idea and the reason why we went this way.

DottoreTozzi, 2018-08-23:
Alright, I've set up a little benchmark.

It consists of a Python Script that sends a typical iSpindle message over TCP via my WLAN Repeater in the basement, through another repeater on ground floor, to the  router on 1st floor, to my Raspi here on 2nd floor, connected via WiFi to that router one floor downstairs.
So this is in many aspects a worst case scenario for both iSpindle and the TCP Backend.

Behold, the benchmark results are in **microseconds**.
Due to me testing a worst-case-scenario here, the time to establish the connection is by far the determining factor.
In a one-step setup (iSpindle connected to the same WiFi as the backend, and no repeaters involved, the whole thing happens within only 200 to 300 microseconds.

Of course the results are not always the same, but here is a typical (average) one, with a full implicit return JSON:

          Received data: { "interval" : 60,"name":"iSpindel01","vfact":191.80,"token":"*dummy","polyn":"0.002378287*tilt^2+0.327188119*tilt-8.88"}
          Time elapsed (in microseconds):
          ----open:  302.0
          ----sent:  54.0
          ----recv:  64.0
          ----close: 72.0
          ----total: 492.0

And here's the same with simple "ACK" reply and the sender (iSpindel) discarding the return value:

          Time elapsed (in microseconds):
          ----open:  310.0
          ----sent:  58.0
          ----total: 453.0

So, I think it's absolutely safe to say that the payload regarding implicit configuration changes is completely neglectible and takes a fraction of the costs an explicit call over http would have.
Quod erat demonstrandum.

DottoreTozzi, 2018-08-23:
Alright, the results from iSpindle itself are in and completely as expected, as well, since we're in the **micro** second realm here as I have established in the previous post. These differences are completely within random environment interference limits and would even, if anything, imply the contrary of earlier assumptions (by random, of course).
Truth is, we can't assume **any** difference **at all**.

**Full JSON (all useful parameters included):**
calling TCP
{"name":"iSpindle01","ID":13582305,"token":"*dummy","angle":67.18478,"temperature":30.4375,"battery":4.07195,"gravity":0,"interval":60,"RSSI":-34}Sender: TCP stream
{ "interval" : 60,"name":"iSpindel01","vfact":191.80,"token":"*dummy","polyn":"0.002378287*tilt^2+0.327188119*tilt-8.88"}
Final-sleep: 60s; **RT:3471**

**JSON containing only Interval:**
calling TCP
{"name":"iSpindle01","ID":13582305,"token":"*dummy","angle":67.02964,"temperature":30.8125,"battery":4.07195,"gravity":0,"interval":60,"RSSI":-35}Sender: TCP stream
{ "interval" : 60 }
Final-sleep: 60s; **RT:3470**

**ACK 1-byte-handshake only:**
calling TCP
{"name":"iSpindle01","ID":13582305,"token":"*dummy","angle":67.19608,"temperature":30.5,"battery":4.07195,"gravity":0,"interval":60,"RSSI":-36}Sender: TCP stream
Final-sleep: 60s; **RT:3473**

So yea, by these results (which I anticipated) I think it's safe to say that an implicit config call generates no payload **at all**.
Try and beat that with http. SCNR.

ckrack, 2018-08-23:
Thanks for the benchmarks.
However if your point is to prove that TCP is superior in this use-case, this is not much help.

In reality people just use it.
@tgangso does with brewfather, the craftbeerpi plugin does, monitor.beer does, brewstat.us does..
Some have options, some don't.
I do support both.

To validate the assumption that it's fine to use implicit configuration, we need a HTTP benchmark, too, and we need to compare these times with the config-parsing inside the spindel firmware to see whether it's relevant at all.

DottoreTozzi, 2018-08-23:
Nope. You are unfortunately still completely missing the point, apparently.
We are talking about a backend inside the home network here which can (and usually will) work as a relay to the oh-so-dangerous outside world, i.e. all those bad thugs who want to steal and manipulate my fermentation data.
In this case (and this case only), Payload generated by a TCP reply is completely irrelevant.
This is what remained to be demonstrated and now has been.

HTTP will be ideal to forward data thus gathered, **after** the connection to the IoT device has been severed and the data processed.
This can be done by a simple authenticated GET.
So, any other solutions, be it Ubidots, or yours, I won't discuss, as this would mean trying to cover a scope completely different from what I am saying here.

Take a look at the [OSI Model](https://en.wikipedia.org/wiki/OSI_model) in order to see what I mean.
An Application Level protocol is completely unneccessary for our purposes here and introduces overhead that can otherwise be avoided.

Please try to respect my work for what it is, as mine is a completely separate construction site, having nothing to do with yours (as you assured earlier).

If you wish to implement another way to enable remote configuration of the iSpindel, please open your own issue here on github and stop spamming mine, as I will abstain from further commenting your work on this.

In case you wish to discuss any of this, let's take it elsewhere and keep Sam's Repo clean.

Thanks.

ckrack, 2018-08-24:
@DottoreTozzi 
You started this issue with:
> We should discuss what a remote configuration of the iSpindle from a local or public server might look like.
So that is what we are about. Not limited to any implementation, we are discussing a way to make remote configuration work for everyone.

Sam reacted to this request by adding experimental support for getting the interval solely and with support for ubidots solely. So he does not seem to agree that this issue is about your implementation of a TCP server only, either.
I would like to generalize this implementation so it works for and with any endpoint, including the existing ones that i named.

We can go on in different, separated issues if you like, all it takes would be to close this and open one that is for the (local) TCP implementation only.
But in my opinion we should keep this issue as a place where everyone can discuss what should be done about the general implementation of a remote configuration, as initially stated by you.
If you don't want to receive email notifications, you can also mute the thread (link is in every email sent by github).
Let's keep this on a productive level.

Misenkooo, 2018-10-12:
@universam1 Hi, I wanted to ask how does it look like with remote config from Ubidots? I saw the remote branch is not maintained anymore and remote functionality is not merged into the master branch. Is there any reason for that ? Would be great if there is some remote control possible from UD. Thank you.

universam1, 2018-10-14:
Before I merge it I wanted to get more feedback. Let me know how it works for you

Misenkooo, 2018-11-25:
I've abandoned ubidots and moved to TCP communication. But here it does not work at all even though generic TCP server supports remote config. I can see server send the response but it's not processed by iSpindel firmware. Maybe you could have a look ? If I have time maybe I could look into it as well. Thanks.

mcmellow, 2020-02-21:
Nothing added since 25 Nov 2018.  Any progress in Remote Configuration Support?

closed 2020-06-23 by stale[bot]

ckrack, 2020-06-23:
@universam1 Is this really supposed to be closed?
There is already some code to handle this, and it would be great to get this fixed.

mcmellow, 2020-06-23:
I would be very happy to change some of the configuration "on the fly" while it is active a the fermenting process.

pe1rmg, 2023-10-22:
I am using an MQTT setup with Home Assistant. I would love to have the option to request via MQTT the iSpindel to return to config AP mode? In essence just check if a certain topic (like "configrequest") is available with a value != 0. If it reads the topic as != 0 reset the iSpindel in config AP mode. Then during the writing of the config, also write this topic back to 0.

I am not familiar enough with PIO, and am unable to get the development environment setup. (It can not locate / download certain libs). Otherwise I would give it a try myself to implement.....