universam1/iSpindel #97: iSpindel forgets SSID/password

herrpaul, 2017-09-20:
Whenever entering configuration mode, the fields for SSID and password are blank and have to be re-entered again, otherwise iSpindel cannot send data in operation mode.

If there are no security objections (after all, anybody can access iSpindel when it is in configuration mode) it would be nice if the SSID/password are stored. Maybe both, but at least the password should be shown with stars ("***") only, not allowing to read it but only to overwrite it.

universam1, 2017-09-20:
While it does not forget the connection settings, it currently does not display them again. 
I'm okay to add this as a todo item

herrpaul, 2017-09-20:
Hm, I'm pretty sure that I tried the following at least once:

Enter configuration mode
Leave (empty) SSID/passwort fields alone
Changed intervall
Pressed Save to go to operation mode
-> no more data received on Ubidots

That's why I was pretty sure that the data is not just invisible, but actuall gone. It was my first day with the iSpindel, though, so something else might have caused this behaviour. After all you know your code :)

Otherwise, your proposed fix looks fine for me, thanks.

universam1, 2017-09-21:
My iSpindel is currently making beer for me, so I can't test. Here is the version which should redisplay already saved credentials.

Please test it out and feed back.
[firmware.zip](https://github.com/universam1/iSpindel/files/1321289/firmware.zip)

herrpaul, 2017-09-21:
Mine is at work right now, too :) So testing will have to wait.

universam1, 2017-09-28:
deployed with 5.5.0

closed 2017-09-28 by universam1