universam1/iSpindel #100: Missing data sending FHEM (since firmware 5.4.2)

cwilling, 2017-10-05:
For both 5.4.2.and 5.5.0, my server is receiving data like:

    Request for /fhem received from ::ffff:10.1.1.132
    Request query = cmd.Test=set%20iSpindelKB00%20%20%20%20&XHR=1

i.e. correct device name but all other fields empty.

Previous versions 5.1.2 & 5.2.3 sent data correctly e.g.

    Request for /fhem received from ::ffff:10.1.1.132
    Request query = cmd.Test=set%20iSpindelKB00%2082.70%2029.56%204.80%2029.89&XHR=1

i.e. data fields contain some values.

universam1, 2017-10-06:
please post the console output of the iSpindel here as well, it is critical to see what parts are actually rendered and where the data gets lost. In the source code is no obvious bug and I can't test it here.

universam1, 2017-10-06:
[firmware.zip](https://github.com/universam1/iSpindel/files/1361695/firmware.zip)

Here is a firmware with more debug enabled for FHEM, please flash and post the console output

cwilling, 2017-10-06:
Output as requested:

FW 5.5.1
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindelKB00","Token":"","Sleep":20,"Server":"10.1.1.13","API":5,"Port":8888,"URL":"","Vfact":191.8,"SSID":"Budiiprimary362AEB","PSK":"ad9c7316","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 87.51
Spl 1: 87.47
Spl 2: 87.53
Spl 3: 87.47
Spl 4: 87.49
Spl 5: 87.41
Spl 6: 87.49

a: -15806	698	76	absTilt: 87.49	T: 33.20	V: 4.80	owT: 30.37	Gravity: 32.30
After waiting 0.00 s, result 3
10.1.1.132

calling FHEM
{"angle":87.4864,"temperature":30.375,"battery":4.801877,"gravity":32.30322}Sender: FHEM get
DEBUG: FHEM data:

end DEBUG: FHEM data
GET /fhem?cmd.Test=set%20iSpindelKB00%20%20%20%20&XHR=1 HTTP/1.1
Host: 10.1.1.13:8888
User-Agent: iSpindelKB00
Connection: close

Final-sleep: 20s; RT:3929

universam1, 2017-10-06:
Wonderful, thank you!

It looks like there is a bug in ArduinoJSON. Fortunately, I think I found a way around it by using a different method.

Can you please try this new version and report back:
[firmware.zip](https://github.com/universam1/iSpindel/files/1361982/firmware.zip)

cwilling, 2017-10-06:
Yep, looks good now (output below). Thanks for the fast fix.

FW 5.5.2
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindelKB00","Token":"","Sleep":20,"Server":"10.1.1.13","API":5,"Port":8888,"URL":"","Vfact":191.8,"SSID":"Budiiprimary362AEB","PSK":"XXXX","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 83.64
Spl 1: 83.63
Spl 2: 83.72
Spl 3: 83.75
Spl 4: 83.68
Spl 5: 83.69
Spl 6: 83.64

a: -15718	-1478	1028	absTilt: 83.67	T: 32.57	V: 4.80	owT: 29.94	Gravity: 30.38
After waiting 0.00 s, result 3
10.1.1.132

calling FHEM
{"angle":83.66948,"temperature":29.9375,"battery":4.801877,"gravity":30.37972}
Sender: FHEM get
DEBUG: FHEM data:
83.66948
29.9375
4.801877
30.37972
end DEBUG: FHEM data
GET /fhem?cmd.Test=set%20iSpindelKB00%2083.66948%2029.9375%204.801877%2030.37972&XHR=1 HTTP/1.1
Host: 10.1.1.13:8888
User-Agent: iSpindelKB00
Connection: close

HTTP/1.1 200 OK
Content-Type: image/x-icon
Date: Fri, 06 Oct 2017 09:30:08 GMT
Connection: close
Transfer-Encoding: chunked

0

Final-sleep: 20s; RT:1928

closed 2017-10-06 by cwilling