universam1/iSpindel #101: Sending http body before headers

j0hnsmith, 2017-10-05:
[image]

Http post requests aren't being registered by BrewPiLess (bottom shell), so I made a request manually via curl which worked (the one request you can see that did work). I realised that the problem is iSpindel (top  shell) is sending the body before the headers!

universam1, 2017-10-06:
> I think this is just displaying the body before the headers in the serial
> output. I tried many different versions of iSpindel and *none* of them
> worked with BrewPiLess (but did display the correct http in the serial
> output).
>
> Eventually I got it working with the iSpindel.bin
> in BrewPiLess/extra. Not sure what's different about that version.

Can you please post the output of the older firmware here as well, maybe we
can diff the reason why BrewPiLess is not accepting the data now.

j0hnsmith, 2017-10-06:
@universam1 I've created #102 with lots of detail and videos that show the problem