universam1/iSpindel #102: iSpindel POST to BrewPiLess fails

j0hnsmith, 2017-10-06:
So I've made some videos to demonstrate (click the images to download mp4 videos)

### Shells
top: curl
middle: iSpindel (various versions)
bottom: BrewPiLess (f7b3bf1)

### iSpindel 5.5.0
The first shows iSpindel 5.5.0, the http body is appearing before the headers but maybe that's just the serial output. Note that BPL is showing `BodyStart: ... B` but the request never seems to finish. When I make a request via curl it works as expected

[image]

### iSpindel 5.1.2
This shows the http body after the headers, BPL is showing `BodyStart: ... B` but again the request never seems to finish. With curl it works as expected but after a few requests BPL seems to freeze, the second curl request doesn't connect. I think the freeze is issue [referenced in the BPL readme that's now been resolved](https://github.com/vitotai/BrewPiLess#ispindel-support).

[image]

### iSpindel 5.1.2 ([BPL extra](https://github.com/vitotai/BrewPiLess/blob/master/extra/iSpindel-BPL.5.12.bin))
This is the only version that I've been able to get working, I'm not sure what is different in this version. Note that the http POST request takes around 13 seconds to complete, that is a _very_ long request. The curl request seems to take a fraction of a second. 

@vitotai What's different about [this](https://github.com/vitotai/BrewPiLess/blob/master/extra/iSpindel-BPL.5.12.bin) version of iSpindel?

[image]

## Bug
It seems odd that the POST request should take more than say 250ms, I'm guessing the fact that it takes ~13 seconds is part of the problem, where it's not working I suspect something is timing out before the request is complete. However the big question is why does it take so long in the first place?

@vitotai Do you have any idea why this might be happening?

universam1, 2017-10-08:
Thanks for the feedback!
Here is a bugfixed version, using a library now to perform the post. Please also monitor the execution time in this regard and let me know how it works.
[firmware.zip](https://github.com/universam1/iSpindel/files/1366844/firmware.zip)

universam1, 2017-10-08:
duplication of #103

closed 2017-10-08 by universam1

j0hnsmith, 2017-10-08:
Great news, my iSpindel & BPL are otherwise engaged (in a brew) so it'll be a while before I can test and report back.