universam1/iSpindel #106: Container options

vytux-com, 2017-10-09:
I am struggling to find a decent source of PET Preforms as required for this project. I did manage to get hold of a PET bottle with the same diameter top and only slightly longer than the preform. However it is a bit wider at the bottom and has the standard bottle flared bottom.
[image]
Would it be possible to use this instead and are there any requirements for getting this to tilt correctly?

universam1, 2017-10-09:
This is hard to say if it will work or not, you are basically on your own to try. 
However, you should be able to buy a PET preform which is hard to find but usually available through geo caching shops.

vytux-com, 2017-10-09:
Unfortunately the only ones I could find were in Europe and shipping them half way around the globe makes them way too expensive.

If the container was 25mm or so too long would you extend the drawer by that amount or would you pack something in the bottom of the bottle?

Philusha1983, 2017-10-09:
http://www.ebay.com/itm/132339878510?ssPageName=STRK:MESELX:IT&_trksid=p3984.m1555.l2649

bralexc, 2017-10-09:
You don't _have_ to use the 3D printed drawer from the project. After you've selected a bottle, you can make your own drawer with any light material that is easy to sculp, like balsa wood or styrofoam. Just be sure that the components don't juggle around once you've mounted everything. You might also have to play a little with the extra weights to get the CG in place.

filipematos95, 2017-11-13:
> http://www.ebay.com/itm/132339878510?ssPageName=STRK:MESELX:IT&_trksid=p3984.m1555.l2649

 @Philusha1983 the PET container from that link only have a diameter of 33 mm, there are no drawers for that diameter.

Philusha1983, 2017-11-13:
> MESELX:IT&_trksid=p3984.m1555.l2649

This is the correct link.
I bought it and used the pet to make several ispendels

filipematos95, 2017-11-13:
@Philusha1983 which .stl file did you use for those PET?

Philusha1983, 2017-11-13:
https://github.com/universam1/iSpindel/blob/master/drawer/drawer-combo-short-spring2.stl

universam1, 2017-11-17:
assume it is fine, so closing for now

closed 2017-11-17 by universam1