universam1/iSpindel #109: V3 of the wemos just arrived

charliwest, 2017-10-12:
Anyone able to confirm if the newest version works?

universam1, 2017-10-12:
Have some links?

charliwest, 2017-10-12:
http://s.aliexpress.com/fmAvuEju

zoomx, 2017-10-16:
It seems that the only hardware modification is the jumper fo deep sleep (GPIO16 to reset pin).

charliwest, 2017-10-16:
Ive got some one the way so can hopefully confirm in nov if it works ;)

maramu, 2017-11-05:
Hello,
I bought the version [V2.3.0 D1 mini](https://wiki.wemos.cc/products:d1:d1_mini), I would like to know if I had to buy a new wemos (older version) or If I can use this version. Moreover, I do not know where is the diode. Please, help me!
Thank you so much,
Martín

universam1, 2017-11-06:
not tested myself, but it looks ok to go. The diode is on the same spot as before and has a printing of `SL` on.

charliwest, 2017-11-06:
the 2.3 just moved the ports to the other side for space saving, everything else is the same I believe.

Mvanmierlo, 2017-11-06:
Some pictures as a reference

[image]

[image]

maramu, 2017-11-06:
Thank you very much!.

universam1, 2017-11-17:
assume works all good, so closing for now

closed 2017-11-17 by universam1