universam1/iSpindel #110: MQTT Erweiterung

legatus85, 2017-10-25:
Hallo Leute,
ich würde mich gerne an der iSpindel beteiligen. Es ist für mich auch das erste GitHub Projekt, an dem ich mich beteilige, deswegen würde ich mich über Hilfe/Unterstützung/Milde für einen Unerfahren Padawan wie mich freuen. 

Meine Idee: die mir im Kopf rumspukt ist eine MQTT Erweiterung. Dann könnte man die Bierdaten noch einfacher auf dem eigenen Server speichern. 
Beispiel (an Luftdruck und Temperatur)
[https://screenzone.eu/arduino-mqtt-sensor-grafana/](url)

universam1, 2017-10-25:
Hi!
Ich habe auch schon einiges mit MQTT realisiert und finde es auch sehr praktisch. Ein grosser Nachteil in diesem Zusammenhang ist, dass MQTT sich erst beim Broker anmelden muss. Das passt batterielaufzeittechnisch mir überhaupt nicht - sie muss aufwachen, die Daten absetzen und abtauchen.

Laut einer Hochrechnung lohnt sich MQTT erst ab 5s Verbindungsdauer, ich schaffe es mittlerweile unter 2s. Eine weitere Sekunde bedeutet 1/3 weniger Batterielaufzeit. 
Klar kann man das implementieren, ob es sich aber lohnt?

legatus85, 2017-10-26:
Hey,
der Hinweis ist echt super, danke dafür. 
ich habe eben mal ein paar Untersuchungen der Laufzeit an dem Code aus meinem Link unternommen und folgende Ergebnisse bekommen:
Gesamtlaufaufzeit: 3.36
MQTT Verbindung: 0.05
MQTT Post: 0.04
Wifi Verbindungsaufbau: 3.00
Dazu muss man natürlich sagen, dass mein Server sich im gleichen Netzwerk befindet. Man müsste mal untersuchen wie die Laufzeiten untersuchen, wenn der Broker nicht im gleichen Netzwerk ist. Jedoch sehe ich die Idee von MQTT in diesem Zusammenhang auch mehr, dass man die Daten auf eigenen Geräten bleiben (Beispielsweise ein Raspberry Pi). 
Ich poste einfach mal den Code den ich benutzt habe für meine Untersuchungen. 
Die pubsub library habe ich von @knolleary :  https://github.com/knolleary/pubsubclient

```
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <Adafruit_BMP085.h>

Adafruit_BMP085 bmp;

const char* ssid = "";
const char* password = "";
const char* mqttServer = "192.168.178.2";
const int sleepTimeS = 60;

WiFiClient espClient;
PubSubClient client(espClient);

char valTemperature[5];
char valPressure[5];
long mstart;
long mqttdiff;
long mqttpost;
long wifidiff;

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);

  Serial.begin(115200);
  setup_wifi();

  

  if (!bmp.begin()) {
    Serial.println("BMP085 not found");
    while (1) {}
  }
  mstart = millis();
  client.setServer(mqttServer, 1883);
  
}

void setup_wifi() {
  long wifistart = millis();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  wifidiff = millis() - wifistart;
  
}

void reconnect() {
  long mqttstart = millis();
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    if (client.connect("arduino", "arduino", "arduino")) {
      Serial.println("connected");
      client.publish("sensors/esp2", "hello world");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
  mqttdiff = millis() - mqttstart;
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  // read temperature
  dtostrf(bmp.readTemperature(), 3, 1, valTemperature);
  Serial.println(valTemperature);
  long poststart = millis();
  String payload;
  payload = "sensor2 temperature=";
  payload += valTemperature;
  // publish temperature
  client.publish("sensors/node2/temperature", (char*) payload.c_str());
  // read pressure
  dtostrf(bmp.readPressure(), 3, 1, valPressure);
  payload = "sensor2 pressure=";
  payload += valPressure;
  // publish pressure
  client.publish("sensors/node2/pressure", (char*) payload.c_str());
  mqttpost = millis() - poststart;

  // fall into sleep
  long diff = millis() - mstart;
  Serial.print("Laufzeit: ");
  Serial.println((float)millis()/1000);
  Serial.print("MQTT Verbindung: ");
  Serial.println((float)mqttdiff/1000);
  Serial.print("MQTT Post: ");
  Serial.println((float)mqttpost/1000);
  Serial.print("Wifi Verbindungsaufbau: ");
  Serial.println((float)wifidiff/1000);
  ESP.deepSleep(sleepTimeS * 1000000);
}
```

universam1, 2017-10-26:
Es ist richtig dass die meiste Zeit für den Wlan Verbindungsaufbau und DHCP drauf geht. Wobei das bei dir schon sehr langsam ist, meine Messungen sind meistens ~1800ms.
Cloud upload (Ubidots) ist schon lang keine Einschränkung mehr, siehe Raspi server, FHEM, Openhab und vor allem die generic HTTP / TCP Schnittstelle. Schneller als letztere wird es auf keinen Fall.

Aber wenn du dir die Mühe machen willst eine MQTT Schnittstelle, analog zu den vorhandenen zu coden und als PR bereitstellst, nehme ich das sehr gerne mit auf!

universam1, 2017-12-14:
Any activity here?

PjotrekSE, 2017-12-25:
Maybe a good idea to implement the MQTT part in the iSpindel-TCP-Server?

universam1, 2018-02-21:
Could be an interesting option. No activity here so makes no sense to keep it open

closed 2018-02-21 by universam1

jmelhus, 2018-08-08:
I've implemented MQTT binding in iSpindel with the use of PubSubClient. Works great with openHAB. I've forked the project here: https://github.com/jmelhus/iSpindel , and made a pull request here: https://github.com/universam1/iSpindel/pulls

universam1, 2018-08-08:
Great, looks good. Two remarks in the PR