universam1/iSpindel #111: Precision of transmitted chipId

cwilling, 2017-10-30:
I'm using http protocol.
For a chipId=16611943 (according "Information" at 192.168.4.1) my server is receiving "ID":1.661194e7 which results in 16611940.
For another chipId=16611813, the server receives "ID":1.661181e7, resulting in 16611810.
Is this likely to be the precision transmitted or is it an inherent limitation of the node.js server receiving the transmissions? My server's request handler (before any processing) reports, for example,

    Request handler 'iSpindel' was called with query: {"name":"iSpindelKB00","ID":1.661194e7,"token":"xyz","angle":83.77038,"temperature":28.25,"battery":4.301355,"gravity":30.43068}

Any ideas?

cwilling, 2017-10-30:
PR #112 converts result of getChipId() to a String so there is now no scientific notation precision to worry about in the transmission to server. The server request handler now reports:

    Request handler 'iSpindel' was called with query: {"name":"iSpindelKB00","ID":"16611943","token":"xyz","angle":75.94956,"temperature":28.3125,"battery":4.228363,"gravity":26.46172}

i.e. the ID field is now correct.

The only possible (small) drawback I see is non-compatibility with existing server code - code expecting a number rather than string. My code doesn't care - it's only using the ID field like a string anyway, not doing any calculation with it. I suspect any other code using the ID field is also using it the same way.

universam1, 2017-10-30:
Thanks for reporting!

Unfortunately, type changing to string will break other server implementation, and it is actually a integer not a string.
Looks like Arduinojson is now forcing to scientific notation which is bad.
The proper solution would then be to create a integer function overload that fits better.

PR #113 (2017-10-31): Overload add for uint32_t (as returned by getChipId())

Replaces #112 and fixes #111 correctly (I think).
closed 2017-10-31 by universam1