universam1/iSpindel #118: configuration

maramu, 2017-11-20:
Hello,
I just have finished the construction of the spindel. I have checked using putty and Arduino the connection at 115200 baudio and this is the information (I have hidden with asterisks the name of the wifi networks):

scan start
scan done
10 networks found
ESP8266 Chip id = 003C9793
1: v*********0 (-85)*
2: v**********9 (-88)*
3: c********s (-54)*
4: _A********i (-54) 
5: _O******i (-54) 
6: O******0 (-89)*
7: v*******8 (-92)*
8: _A****i (-91)*
9: _O******i (-90) 
10: H*******C (-82)*

This information is shows every second. In the FAQ I have read: "Switch it on, press the reset button 3-4 times which sets up an access point", but I do not see any new wifi network in my computer. What could be the problem?, I have not done a previous step ?, What can I do?. I suppose that some drivers must be upload to the wemos but how.
Thank you very much,
Martín

universam1, 2017-11-21:
I guess you did not flash this Wemos with the iSpindel firmware, right? 😊
There is a section in Readme how to do that.

maramu, 2017-11-21:
Perfect!!!!.
Thank you very much.

closed 2017-12-12 by universam1