universam1/iSpindel #119: Remove WEMOS input diode

cwilling, 2017-11-23:
This is just a request for clarification about removing the WEMOS input diode, as mentioned in [this circuit diagram](https://github.com/universam1/iSpindel/blob/master/docs/circuit_diagram_en.md) and issue #47.

Is it correct to say that any danger only occurs when the iSpindel is connected by the USB port? For developers, connection by USB is sometimes more convenient e.g. for faster firmware flashing, but the normal user would not normally use this mode.

A second related question; if WEMOS is connected by USB but no battery attached, is the danger still there? I presume no charging current would be drawn in this case (LiPo  led is green if it has USB power attached with no battery).

universam1, 2017-12-12:
Yes this is correct.

closed 2017-12-12 by universam1

einaruk, 2018-02-26:
As I understand from circuit diagram, even if you do not remove the diode, but be careful to turn OFF the switch when you connect the Wemos to USB, there is no danger of creating overvoltage on the battery, thus no risk. (No risk only, If you are very careful and do not forget to switch off every time you connect the Wemos to USB.)  
Correct?

JensW2000, 2018-02-26:
Nope! If you do not the Wemos diode, and put an USB cable into the Wemos port, then the battery gets uncontrolled 5V. This can cause a battery explosion in worst case. The better cases are possible damage of the Wemos, the TP4056 or both. 

You would like to remove the diode, right? :-)

einaruk, 2018-02-27:
Hi @JensW2000,  
  I guess you missed the prerequisite for my statement. I said, "if you switch off the switch".
If you switch off the switch when connecting to USB, how can you deliver 5V to battery?

Am I missing sth? I do not know the function details of the LiPo charger but will 5V input to +IN pin create any damage?

Thanks