universam1/iSpindel #120: Angle Issue...

dxcoco2, 2017-11-26:
I've bought the kit on-line...
i've solder all the board and connection...after flash the firmware i've exec the calibration and i follow all step with water and sugar but....the angle is the same for every solution density!
0°P -> 76,8
2,5°P -> 76,7
5°P -> 76,8

why?

if i force the Ispindle in vertical position, the angle value is 1,5/2°...the gyroscope is ok!
Thanks

benbroadley88, 2017-11-26:
It looks like your device isn't heavy enough. In plain water (0°p / 1.000 SG) the device should float around 25°. Try adding a bit more weight to it and you should see a greater change in angle as the gravity increases/decreases

dxcoco2, 2017-11-26:
ok i try....thanks!

dxcoco2, 2017-11-26:
I've add a weight and now is ok! This week i perform  a calibration with a sugar solution!
Thanks for your help!

closed 2017-12-02 by dxcoco2