universam1/iSpindel #121: configuration errors

jmreidmj, 2017-11-29:
Hello,

Below is the log from attempt to configure iSpindel for Ubidots.  

The first issue is "ERROR: OW DISCONNECTED" which I think may be caused by a broken temp sensor.

The main issue is it appears that the wifi parameters are not being saved.  In the log you see other parameters but not the wifi parameters.

I have confirmed that correct values are entered for SSID and password in the configuration web page.

Greatly appreciate your help.
John

**iSpindel Log 29-Nov-2017**

WM: Handle root
ERROR: OW DISCONNECTED
WM: 3***M
WM: -39
WM: H***0
WM: -45
WM: F***N
WM: -72
WM: S***S
WM: -88
WM: B***0
WM: -90
WM: F***9
WM: -92
WM: N***7
WM: -92
WM: Sent config page
ERROR: OW DISCONNECTED
ERROR: OW DISCONNECTED
WM: Request redirected to captive portal
WM: Request redirected to captive portal
WM: Handle root
ERROR: OW DISCONNECTED
ERROR: OW DISCONNECTED

WM: Request redirected to captive portal
ERROR: OW DISCONNECTED
ERROR: OW DISCONNECTED
WM: Request redirected to captive portal
ERROR: OW DISCONNECTED
ERROR: OW DISCONNECTED
WM: WiFi save
WM: Parameter
WM: name
WM: iSpindel011
WM: Parameter
WM: sleep
WM: 900
WM: Parameter
WM: vfact
WM: 191.80
WM: Parameter
WM: 
WM: 
WM: Parameter
WM: 
WM: 
WM: Parameter
WM: selAPI
WM: 0
WM: Parameter
WM: token
WM: A1E-KzFWwXwK8ym0X3rgpnk5ZsEKTIgSeq
WM: Parameter
WM: server
WM: 
WM: Parameter
WM: port
WM: 80
WM: Parameter
WM: url
WM: 
WM: Parameter
WM: 
WM: 
WM: Parameter
WM: POLYN
WM: -0.00031*tilt^2+0.557*tilt-14.054
WM: Sent wifi save page
ERROR: OW DISCONNECTED

Exception (9):
epc1=0x40104a4d epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys 
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  00000056 00000001 4010417e 3ffeeb90  
3fffff60:  4022a0c5 3ffef970 00000001 00000000  
3fffff70:  4023e683 00000004 00000000 00000000  
3fffff80:  3fff2ab4 4023e032 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffeaa78 00000000 40249163  
3fffffa0:  3ffeaa78 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

FW 5.6.1
2.1.0(deb1901)
Worker run!
Boot-Mode: 2
mounting FS... mounted!

ERROR config corrupted
...........
ERROR no Wifi credentials

going to Config Mode
reboot RFCAL

JensW2000, 2017-11-29:
The main-issue is, that the DS18B20 is broken or incorrectly wired. As a side effect of the missing sensor the iSpindle ran into an exception and the WiFi credentials are not saved.

Please generate a new Ubidots Token and delete the old one. You’ve posted your current token to the public.

Cheers
Jens

jmreidmj, 2017-11-30:
Thanks, Jens!!

closed 2017-12-01 by jmreidmj