universam1/iSpindel #126: Special characters (&%# etc.) in WiFi passphrase, revisited...

DottoreTozzi, 2017-12-22:
Looks like these can still cause problems.
Entering them in iSpindel config works fine for me; however, when being redisplayed (upon invoking config mode), these special characters are missing from the corresponding input field.
When the user submits his config changes and overlooks this, the iSpindel will therefore not connect anymore.

universam1, 2017-12-25:
I think I see the point. 
The special character would need to be html encoded again to properly display in config page. Good catch!

universam1, 2018-01-21:
can you @DottoreTozzi please check if 5.6.2 fulfilles your request?

DottoreTozzi, 2018-01-22:
Would love to. The switch to platformio is currently taking me some time, however.
Do you have a binary for me meanwhile to do a quick test on this?

universam1, 2018-01-22:
My bad I forgot to push the tag. 
Now check this out https://github.com/universam1/iSpindel/releases/tag/5.6.2
Travis built the bin. Have fun

DottoreTozzi, 2018-01-22:
Works like a charm. Issue can remain closed.
Also I have my dev environment up-to-date now and should be able to compile again.
Been slacking on that recently.
Thanks!

universam1, 2018-01-22:
Cool 😎 thanks for feedback

DottoreTozzi, 2018-02-13:
With the latest update, some of the issues reappeared.
The Ampersand character seems to be causing problems.

For example, 
aB&#62Tf%ljixhufs becomes aB%26%2362Tf%25ljxhufs
upon reconfiguring...

universam1, 2018-02-13:
Looks like the browser is not decoding the encoded string. I am not a web developer so this behavior is not completely obvious to me.

DottoreTozzi, 2018-02-13:
Weird. Because it did work. I'll try and investigate.

universam1, 2018-02-14:
Ah understand it now. There are special character that need to be escaped (ampersand encoding) which is different than url encoding. Looks like I need to implement that manually, no routine found for.

DottoreTozzi, 2018-02-14:
The fun thing is that it did work at some point. But your approach to killing this problem so that all browsers will catch it correctly certainly will be the golden path.
It's always these little things that take most time. If there is any way I can assist, please let me know!

universam1, 2018-02-14:
Thanks!

Now I am really convinced that I solved it entirely, introduced a new routine to html encode the character. Did a number of tests with your pattern and not only the PW but all forms display correctly now.
Please check 5.8.1 and let me know!

DottoreTozzi, 2018-02-14:
Yep. That's it. I guess this can be closed for good now!

commit a86f0d2e (2018-01-21): html encode psk to allow special characters
close #126
closed 2018-02-21 by universam1