universam1/iSpindel #127: Suggestion: Use iSpindel + chipid

PjotrekSE, 2017-12-25:
Now the SSID for the configuration of iSpindel uses SSID 'ESP'+chipid. This may be confusing, since many other ESP applications use the same. Better use 'iSpindel'+chipid to uniquely identify the SSID as belonging to an iSpindel in configuration mode.
This would require some minor modifications to WiFiManagerKT.cpp, which is anyway modified for this project.

universam1, 2017-12-25:
How do you come to this conclusion? So actually an iSpindel in config mode has the SSID called `iSpindel`.

PjotrekSE, 2017-12-25:
My error. Must be all the x-mas food. I'm sorry, you are right. Please delete this embarrasing non-issue.
What I really wanted to have changed was the hostname appearing in the router...
Sorry.

closed 2017-12-26 by universam1