universam1/iSpindel #128: Krausen modifies the inclination

dbravoGit, 2017-12-26:
Good morning.

The Spindel, takes correct measurements with dilutions of sugar. But at the time of adding it to the fermenter, it gives me erroneous measurements that I think are due to the formation of Krausen that modifies the inclination of the Spindel.

Had it happened to him or had he thought about it? Any solution?

A greeting

universam1, 2017-12-26:
The CO2 creates a deviation. 
You can cope with that by calibrating this into the equation. 

It has been shown that the best method to create a polynomial is to actually measure the references with a fermentation. 

Hence, see the proposed method how to derive the polynomial in the calibration Readme file.

dbravoGit, 2017-12-26:
Thank you very much, I really use "Easy method (I)", I will do the test with "Improved method using fast fermenting (II)" thank you very much for the information, in the same way I think it will improve the measurement but it will not be exact. I will report the result, thank you very much again.

closed 2018-01-21 by universam1