universam1/iSpindel #131: iOS 11.2.2 and Safari 11.0.2 (Mac OS X High Sierra) break compatibility w/ config mode

DottoreTozzi, 2018-01-20:
Annoyingly, the latest Safari versions seem to no longer work with iSpindel config mode.
Connection with the iSpindel's AP will seemingly be established, but no window opens and access to 192.168.4.1 won't work either.

This breaks configurability with all iOS devices and as such is a major problem.

This might still be a weird problem on my side though, so please try and confirm elsewhere.
And let me know if you want me to look deeper inside this.
I might also open an issue with Apple (I'm a registered developer with them) if it turns out to be a mistake on their part.

vytux-com, 2018-01-20:
Have you tried a different browser, something like Chrome or Firefox? I always thought safari was a bit like IE, not used by anyone wanting stuff to work???

universam1, 2018-01-20:
Just did a test with iOS 11.2.2 and iSpindel Firmware 5.6.1
Works with inbuilt browser and Safari. The dialogue is opened automatically. 

What I realized is that once the network “iSpindel” has been saved it will no longer open the captive portal, that is a iOS inbuilt behavior. Apparently Apple thinks that is convenient for the people. 
The solution is to “forget” this SSID under its settings to retrigger the dialogue. 
Let me know if it works for you

DottoreTozzi, 2018-01-20:
I don't see any way to make iOS "forget" about the network SSID?
However I was able to open the config page under 192.168.4.1 when "connect automatically to this network" is disabled.
No luck with Mac OS X however. And no, using Chrome or Firefox doesn't help.
(Safari is the fastest and safest browser out there at the moment, btw, and without alternative on iOS devices, but that's got nothing to do with the problem at hand).

I am going to look deeper into this issue and will keep you in the loop...

universam1, 2018-01-21:
@DottoreTozzi this explains how to forget the network: http://www.macinstruct.com/node/539

Thats the only way I know of to provoke iOS to pop up the captive portal again.

closed 2018-01-21 by universam1

DottoreTozzi, 2018-01-21:
Uh oh OK. Thanks!
In German it says "Dieses Netzwerk ignorieren", which I was afraid of being permanent...
Makes sense now and solves the problem.
You probably should mention this in the documentation, since I guess more people will fall for this trap eventually...