universam1/iSpindel #133: Question/issue

tedelm, 2018-01-29:
How do I reset the "calibrate" if used i maintenance menu?

I have noticed that when this funktion is used, the tilt value starts to get higher at ~60° tilt. From 80 to 60 its fine but then it starts to increase again.

universam1, 2018-01-30:
You can calibrate any time again if not successfully performed. Lay it flat and don't move during process as shown by the pictogram.
Apparently I don't understand what you mean with tilt goes higher when it gets higher...

tedelm, 2018-01-30:
Re-flashing the firmware doesnt seem to reset the calibration

tedelm, 2018-01-30:
I read thru the code. Could you please insert a button to reset the "my_****" variables used for mpu calibration?

universam1, 2018-01-30:
It’s saved along with the other settings in the config file. You have a hardware issue or you are somewhat performing the calibration in a different way. 
Easiest to reset is to flash a firmware that does a spiffs formatting, I think there is even an Arduino example of that ready available

tedelm, 2018-01-31:
Not tested yet, but here is a .bin file that will format the SPIFFS/CFGFILE.... https://github.com/tedelm/MRTEEL/tree/master/BIN/FormatSPIFFS

closed 2018-01-31 by tedelm

universam1, 2018-02-03:
@tedelm check out new FW v. 5.7.0 where I introduced a `Factory Reset` feature if that fits your needs.

tedelm, 2018-02-03:
Nice work, thanks!